dart-archive/sdk #145: Make the persistent process take the .packages file next to it if none has been provided

ricowind, 2015-09-23:
This will allow for easy bundling of a default .packages file in the initial sdk bundle

sgjesse, 2015-09-23:
What does "next to it" mean? Shouldn't it do this:

"The Dart runtimes, such as the VM, dartjs, and analyzer, will traverse up directories, from the entry point script, to find a .packages. Thus, you do not need a .packages next to each script that you need to run or compile."

(from http://blog.sethladd.com/2015/08/new-dart-sdk-eliminates-symlinks.html).

peter-ahe-google, 2015-09-23:
Consider this situation:

The user installs the Fletch SDK in /usr/local/fletch-sdk.

The user adds /usr/local/fletch-sdk/bin to their PATH.

The user run:

```console
$ fletch run fun_with_gpio.dart
```

In that case we just want to use the default .packages from the SDK that tells fletchc where to find the gpio package.

This may be a short-term thing.

sgjesse, 2015-09-23:
So we supply a default .packages for resolving all packages in pkg that will be used no matter where your Dart file is.

Sounds good to me.

General support for .packages is a separate issue then.

peter-ahe-google, 2015-09-23:
Proposed fix: [CL 1367723002](https://codereview.chromium.org/1367723002/)

With this, the persistent process will look for a file named `fletch-sdk.packages` next to the Dart VM running the persistent process.

I would have liked to make it relative to `driver_main.dart`, but that is blocked on [Dart issue 23951](https://github.com/dart-lang/sdk/issues/23951).

commit c8daac33 (2015-09-24): Fallback package config for Fletch SDK packages

Fixes #145

R=ricow@google.com

Review URL: https://codereview.chromium.org/1367723002 .
closed 2015-09-24 by peter-ahe-google