dart-archive/sdk #148: compiler seems to insert compile-time-errors even if they are not executed at runtime

mkustermann, 2015-09-24:
This program
```dart
main() {
  if (false) {
    a b c ;
  }
}
```
will result in a compile-time error reported when running the program. I would expect that it runs just fine.

By extracting the malicious code sequence into another function, it seems to work. e.g. this program runs fine:

```dart
main() {
  if (false) {
    f();
  }
}

f() {
  a b c;
}
```

cc @peter-ahe-google @skabet

madsager, 2015-09-24:
I think the main method is marked by the compiler frontend as having a compile-time error. When we compile main we therefore just generate a compile-time error. That is also the reason why it works when you extract it. Then main itself is not marked as having a compile-time error and we compile main normally and never get to compile f which contains the error.

cc @johnniwinther

johnniwinther, 2015-09-24:
It touches on the subject of when (at the latest) to give compile-time errors.

The spec has (Ch. 7):

> Compile-time errors are errors that preclude execution. A compile-time error must be reported by a Dart compiler before the erroneous code is executed.

And the continues informally:

> A Dart implementation has considerable freedom as to when compilation takes place. Modern programming language implementations often interleave compilation and execution, so that compilation of a method may be delayed, e.g., until it is first invoked. Consequently, compile-time errors in a method m may be reported as late as the time of m’s first invocation.

The formal specification allows us the handle the program with the error in main, whereas the informal indicates that we shouldn't.

Currently the frontend is doing the 'strict' approach and some work is required for us to tip-toe into the 'lax' approach.

peter-ahe-google, 2015-09-24:
I agree with Johnni's analysis. I don't think we have any current plans be more lax about compile-time errors, but I suspect this may become a priority when live programming is a reality. @lukechurch may have some thoughts on this.

As Mads mentions, the frontend marks the function main as having a compile-time error, and the very first thing we check when compiling a method is if it has a compile-time error. In that case, we generate the bytecodes for signalling a compile-time error.

We continue to generate the rest of the method body, so it is possible that we could easily switch to being more lax. However, I estimate that it will require us to rewrite quite a few language tests to make sure that the compile-time error actually gets executed.

mit-mit, 2015-12-07:
Closing as by-design.

peter-ahe-google, 2015-12-07:
This is a valid request. I was just explaining why the compiler behaves as it currently does, and saying that we don't have any current plans to change it.