dart-archive/sdk #152: Debugger claims that "true" threw an exception

peter-ahe-google, 2015-09-25:
I'm running tests in a version of the compiler that doesn't support type tests with functions.

Notice below that the debugger claims that "true" threw an exception.

```
FAILED: none-fletchc debug_ia32_clang language/closure_type_test/01
Expected: Pass 
Actual: RuntimeError
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugIA32Clang/fletch-vm &
$ out/DebugIA32Clang/fletch attach tcp_socket 127.0.0.1:62087 in session 5

 => 0
$ out/DebugIA32Clang/fletch run .../fletch/out/DebugIA32Clang/generated_tests/language/closure_type_test_01.dart in session 5
.../fletch/out/DebugIA32Clang/generated_tests/language/closure_type_test_01.dart:16:14:
Warning: 'double' is not assignable to 'int'.
  sqrt(x) => math.sqrt(x);
             ^^^^^^^^^^^^
.../fletch/lib/core/string.dart:34:1:
Hint: Adding is-selector for String
abstract class _StringBase implements String {
^^^^^^^^
.../fletch/lib/core/string.dart:453:1:
Hint: Adding is-selector for _OneByteString
class _OneByteString extends _StringBase {
^^^^^
.../fletch/lib/core/int.dart:7:1:
Hint: Adding is-selector for int
abstract class _IntBase implements int {
^^^^^^^^
.../fletch/lib/core/string.dart:496:1:
Hint: Adding is-selector for _TwoByteString
class _TwoByteString extends _StringBase {
^^^^^
.../dart/sdk/lib/core/bool.dart:14:1:
Hint: Adding is-selector for bool
class bool {
^^^^^
Using custom enqueuer
Uncaught exception:
a Instance: 
  - class = class  - @0 = Expect.equals(expected: <4>, actual: <4.0>) fails.

Stack trace:
> 0: test       @true
  1: main       @test(root,4)

.../fletch/out/DebugIA32Clang/generated_tests/language/closure_type_test_01.dart:42:22
37       // a double at run-time.
38       // Therefore, eliminating the run-time type check would be wrong.
39       int x = func(value);
40       Expect.equals(value, x * x);
41     } on TypeError catch (error) {
42       got_type_error = true;
43     }
44     // Type error expected in checked mode only.
45     Expect.isTrue(got_type_error == isCheckedMode());
46   }
47   
[5: 2015-09-25 13:24:10.526 created session with file://.../fletch/.fletch-settings Settings(packages: file://.../fletch/.packages, options: [], constants: {})]
[5: 2015-09-25 13:24:10.527 Connected to Fletch VM on TCP socket 62088 -> 62087]
[5: 2015-09-25 13:24:10.982 Compiled 'file://.../fletch/out/DebugIA32Clang/generated_tests/language/closure_type_test_01.dart' to 3328 commands]
[5: 2015-09-25 13:24:11.145 Uncaught error]

 => 255
$ out/DebugIA32Clang/fletch x-end session 5
Ended session '5'.

 => 0
$ out/DebugIA32Clang/fletch-vm
Uncaught exception:
a Instance: 
  - class = class  - @0 = Expect.equals(expected: <4>, actual: <4.0>) fails.

Command[fletch_session]: .../fletch/tools/testing/bin/mac/dart -c -Dtest.fletch_session_command.is_incremental_enabled=false tools/testing/dart/fletch_session_command.dart out/DebugIA32Clang/fletch .../fletch/out/DebugIA32Clang/generated_tests/language/closure_type_test_01.dart
OR
gdb -ex 'follow-fork-mode child' -ex run --args out/DebugIA32Clang/dart -c -ppackage/ tests/fletchc/run.dart .../fletch/out/DebugIA32Clang/generated_tests/language/closure_type_test_01.dart
OR
In one terminal:
  gdb -ex run --args out/DebugIA32Clang/fletch --port=54321
In another terminal:
  out/DebugIA32Clang/dart -c -ppackage/ -DattachToVm=54321 tests/fletchc/run.dart .../fletch/out/DebugIA32Clang/generated_tests/language/closure_type_test_01.dart
Took 0:00:00.741698

Short reproduction command (experimental):
    python tools/test.py --kill-persistent-process=0 -rfletchc -t120 language/closure_type_test/01

```

peter-ahe-google, 2015-09-25:
I guess this is a "rethrow" issue. Perhaps this is a VM issue.

madsager, 2015-09-25:
I don't think this is a VM issue. If you run this in the debugger and type 'disasm' you can see the generated bytecode and you can see where we are in the bytecode stream. I'm guessing that our calculation of a user-code position for the rethrow turns out to be the 'true' position. So I think this is a bug in the way we map bytecode index back to source.

peter-ahe-google, 2015-09-25:
Mads, that is correct. However, the stack is cut short as well. You'd want to see where the exception was thrown originally, not where it wasn't caught. So I think this issue relates to both the VM and the compiler.

madsager, 2015-09-25:
I think that is two separate issues. One is that true for the position of the rethrow is not very useful (that is this bug). The other is how we should handle exception throwing. Currently, if there is a rethrow the exception is handled as caught, the stack is unwound and therefore there is currently no way to do anything about the "original" throw location. I agree that adding that would be cool. Maybe that is a separate bug with a longer time frame? :-)

mit-mit, 2015-12-07:
Tagging the position & rethrow issue as a fixit candidate.

kasperl, 2015-12-09:
I think you meant to close issue #153.

herhut-ggl, 2015-12-09:
Yes, indeed. Sorry for that and thanks for fixing it.

commit d522da62 (2015-12-09): Properly handle corner cases of the call getter.

Fixes #152.
Fixes #354.

BUG=152, 354
R=kasperl@google.com

Review URL: https://codereview.chromium.org/1505803003 .
closed 2015-12-11 by madsager