dart-archive/sdk #154: Move .fletch to $HOME

peter-ahe-google, 2015-09-28:
The persistent process should store its "configuration" file in $HOME.

kasperl, 2015-10-01:
Are we going to version the .fletch file so we can automatically shutdown an out-dated persistent process?

peter-ahe-google, 2015-10-01:
We need to do something. Short term we really need to move .fletch to a known location. Using the SDK directory wont work as you're supposed to be able to install it in /usr/.

One possibility is that we include the version number in the .fletch file's name (only for sdk releases, for developer builds, I think we should use a fixed name).

In addition, I'd like to figure out if we can automatically shut down the background process. We probably need separate bugs for all these things (which I can file before I close this bug).

ErikCorryGoogle, 2015-10-01:
Could we shut it down as part of gclient runhooks?  I can't see any situation where you would want to do runhooks and then continue to use the persistent process from before.

And how about when we write a new fletch-vm?

peter-ahe-google, 2015-10-01:
I think this bug is about our users. For us Fletch developers, we already shut down the persistent process as part of running tests.

peter-ahe-google, 2015-10-01:
I've implement this:

First look for $FLETCHRC which will be the name of the configuration (socket file).

If $FLETCHRC isn't defined, use "$HOME/.fletch" (if $HOME isn't defined, use getpwuid_r).

See [CL 1373773005](https://codereview.chromium.org/1373773005).

peter-ahe-google, 2015-10-01:
Changed the name from `FLETCHRC` to `FLETCH_SOCKET_FILE`.

commit 9b4b06e0 (2015-10-01): Move .fletch to ~/.fletch

Fixes #154

R=kustermann@google.com

Review URL: https://codereview.chromium.org/1373773005 .
closed 2015-10-01 by peter-ahe-google