dart-archive/sdk #155: Remove GNUMakefile

peter-ahe-google, 2015-09-28:
I should remove GNUMakefile as soon as possible.

ErikCorryGoogle, 2015-09-29:
Bug reports that talk about removing things without talking about the replacement make me nervous.  However in this case I don't use the Makefile :-)

Can we have a BUILD file at the top level with information on how to build?  The advantages of this over having a (non-trivial) wiki page are:

* It means you have the version of BUILD that corresponds to the version you checked out.
* It enables working offline.
* The update of the instructions can be part of the same code review process as the changes to the scripts that they correspond to.

mkustermann, 2015-09-29:
There is one use case for a Makefile (or an equivalent ninja file) I see: We are getting more configurations, more ifdef guarded code and it becomes increasingly hard to make sure that a change someone tests (maybe just in one configuration) actually works everywhere.

This is also true for building: For example I don't know how to build LK (or embed), but I'm pretty sure that there is a very high chance I might break the build with things I'm working on. So having a Makefile which would include: release, debug, asan, clang, lk, mbed, android configurations would be beneficial to ensure one's change at least doesn't break the build of these.

peter-ahe-google, 2015-09-29:
@mkustermann are you actually using the GNUMakefile?

mkustermann, 2015-09-29:
I use it occasionally, because `make` is much faster to write than `ninja -C out/ReleaseIA32` (yes, yes, it does compile more things then, but it can do that in the background) and it gives a bit more confidence that I can build (most of the) configurations.

Though just for that it's not so useful. If we added a LK, Android, mBed configuration that it would be very valuable to me. [Though it doesn't have to be a Makefile -- just a simple command which makes sure a CL doesn't break any of the configurations we care about (which has happend often in the past, in fact)]

peter-ahe-google, 2015-09-29:
If you're actually using the makefile, I won't remove it. However, I don't plan to maintain it or implement feature requests.

closed 2015-09-29 by peter-ahe-google