dart-archive/sdk #158: Attaching twice to the same Fletch VM gives "Connection refused"

sgjesse, 2015-09-29:
$ fletch/out/DebugX64/fletch attach tcp_socket pi2:47704
$ fletch/out/DebugX64/fletch attach tcp_socket pi2:47704
Unable to establish connection to pi2:47704: Connection refused

Of cause the VM on pi2:47704 cannot handle another connection, so the error is sort of correct. However maybe the persistent process can check it is already have a session and provide a better error message.

peter-ahe-google, 2015-09-29:
I think the what should happen is the following:

```console
$ fletch/out/DebugX64/fletch attach tcp_socket pi2:47704
$ fletch/out/DebugX64/fletch attach tcp_socket pi2:47704
Already connected to Fletch VM on pi2:47704; Try running 'fletch disconnect'
```

@lukechurch WDYT? Is 'fletch disconnect' a thing?

kasperl, 2015-09-29:
detach?

peter-ahe-google, 2015-09-29:
I like `detach`.

peter-ahe-google, 2015-10-01:
As @lukechurch points out, as long as disconnecting from the Fletch VM isn't a reversible operation, we shouldn't call it `detach`.

We should change `shutdown` to shutdown the current session (or all). If no sessions remains after calling shutdown, the persistent process should exit.

peter-ahe-google, 2015-10-01:
Low priority for 0.1.0