dart-archive/sdk #159: fletchc error when running with ./tools/test.py

mkustermann, 2015-09-29:
While running tests, I encountered this error (seems flaky):

```
$ ./tools/test.py -mdebug --clang --asan -ax6
...

FAILED: none-fletchc debug_x64_clang_asan corelib/hash_set_test/none
Expected: RuntimeError 
Actual: Crash
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugX64ClangAsan/fletch-vm &
$ out/DebugX64ClangAsan/fletch attach tcp_socket 127.0.0.1:59428 in session 18

 => 0
$ out/DebugX64ClangAsan/fletch run /usr/local/google/home/kustermann/repositories/fletch-gclient/fletch/out/DebugX64ClangAsan/generat
ed_tests/corelib/hash_set_test_none.dart in session 18
/usr/local/google/home/kustermann/repositories/fletch-gclient/fletch/out/DebugX64ClangAsan/generated_tests/corelib/hash_set_test_none
.dart:319:6:
Internal Error: The compiler crashed when compiling this element.
void main() {
     ^^^^
The Fletch compiler is broken.

When compiling the above element, the compiler crashed. It is not
possible to tell if this is caused by a problem in your program or
not. Regardless, the compiler should not crash.

The Fletch team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dart-lang/fletch/issues/new

Please include the following information:

* the name and version of your operating system

* the entire message you see here (including the full stack trace
  below as well as the source location above)

'package:compiler/src/universe/universe.dart': Failed assertion: line 676 pos 12: 'kind == SelectorKind.OPERATOR ||
           kind == SelectorKind.INDEX ||
           !Elements.isOperatorName(memberName.text) ||
           identical(memberName.text, '??')' is not true.
#0      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:27)
#1      Selector.Selector.internal (package:compiler/src/universe/universe.dart:676:12)
#2      Selector.Selector (package:compiler/src/universe/universe.dart:705:27)
#3      ResolverVisitor.handleLocalAccess (package:compiler/src/resolution/members.dart:2392:22)
#4      ResolverVisitor.handleResolvedSend (package:compiler/src/resolution/members.dart:2578:14)
#5      ResolverVisitor.handleUnqualifiedSend (package:compiler/src/resolution/members.dart:2618:14)
#6      ResolverVisitor.visitSend (package:compiler/src/resolution/members.dart:2631:14)
#7      Send.accept (package:compiler/src/tree/nodes.dart:381:38)
#8      CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#9      ResolverVisitor.resolveArguments (package:compiler/src/resolution/members.dart:804:33)
#10     ResolverVisitor.handleStaticOrTopLevelAccess (package:compiler/src/resolution/members.dart:2440:11)
#11     ResolverVisitor.handleResolvedSend (package:compiler/src/resolution/members.dart:2580:14)
#12     ResolverVisitor.handleUnqualifiedSend (package:compiler/src/resolution/members.dart:2618:14)
#13     ResolverVisitor.visitSend (package:compiler/src/resolution/members.dart:2631:14)
#14     Send.accept (package:compiler/src/tree/nodes.dart:381:38)
#15     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#16     ResolverVisitor.visitReturn (package:compiler/src/resolution/members.dart:3134:5)
#17     Return.accept (package:compiler/src/tree/nodes.dart:1089:38)
#18     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#19     ResolverVisitor.visitFunctionExpression (package:compiler/src/resolution/members.dart:558:5)
#20     FunctionExpression.accept (package:compiler/src/tree/nodes.dart:789:38)
#21     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#22     ResolverVisitor.visitNamedArgument (package:compiler/src/resolution/members.dart:3912:12)
#23     NamedArgument.accept (package:compiler/src/tree/nodes.dart:1655:38)
#24     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#25     ResolverVisitor.resolveArguments (package:compiler/src/resolution/members.dart:804:33)
#26     ResolverVisitor.visitNewExpression (package:compiler/src/resolution/members.dart:3345:25)
#27     NewExpression.accept (package:compiler/src/tree/nodes.dart:526:38)
#28     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#29     ResolverVisitor.visitReturn (package:compiler/src/resolution/members.dart:3134:5)
#30     Return.accept (package:compiler/src/tree/nodes.dart:1089:38)
#31     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#32     ResolverVisitor.visitFunctionExpression (package:compiler/src/resolution/members.dart:558:5)
#33     FunctionExpression.accept (package:compiler/src/tree/nodes.dart:789:38)
#34     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#35     ResolverVisitor.resolveArguments (package:compiler/src/resolution/members.dart:804:33)
#36     ResolverVisitor.handleStaticOrTopLevelAccess (package:compiler/src/resolution/members.dart:2440:11)
#37     ResolverVisitor.handleResolvedSend (package:compiler/src/resolution/members.dart:2580:14)
#38     ResolverVisitor.handleUnqualifiedSend (package:compiler/src/resolution/members.dart:2618:14)
#39     ResolverVisitor.visitSend (package:compiler/src/resolution/members.dart:2631:14)
#40     Send.accept (package:compiler/src/tree/nodes.dart:381:38)
#41     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#42     ResolverVisitor.visitExpressionStatement (package:compiler/src/resolution/members.dart:488:5)
#43     ExpressionStatement.accept (package:compiler/src/tree/nodes.dart:1155:38)
#44     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#45     ResolverVisitor.visitNodeList (package:compiler/src/resolution/members.dart:3106:7)
#46     NodeList.accept (package:compiler/src/tree/nodes.dart:565:38)
#47     CommonResolverVisitor.visit (package:compiler/src/resolution/resolution_common.dart:982:54)
#48     ResolverVisitor.visitIn (package:compiler/src/resolution/members.dart:451:31)
#49     ResolverVisitor.visitBlock (package:compiler/src/resolution/members.dart:471:5)

```

cc: @peter-ahe-google

peter-ahe-google, 2015-09-29:
Did you repeat "corelib/hash_set_test/none" and verify that it was flaky?

mkustermann, 2015-09-29:
Yes I did.

peter-ahe-google, 2015-09-29:
Thank you for confirming. I think this is the same problem that we occasionally see on the build bot.