dart-archive/sdk #162: fletch_tests/incremental/fix_compile_time_error_in_field hang on build bot

peter-ahe-google, 2015-09-30:
Flaky test on build bot.

In addition to the log below, we could observe a hanging fletch-vm process after the build. This process could easily be the one that hung in the feature test.

```
Hanging process info:
  PID: 15846
  Command line: /mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/fletch-vm -Xvalidate-heaps
Killed pid: 15846
```

```
@@@BUILD_STEP test_DebugIA32Clang failures@@@

FAILED: none-fletch_tests debug_ia32_clang fletch_tests/incremental/fix_compile_time_error_in_field
Expected: Pass 
Actual: Timeout
CommandOutput[fletch_test]:

stdout:
Test 'fix_compile_time_error_in_field'
Running '/mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/fletch-vm -Xvalidate-heaps'
Program version 1 #10:
==> main.dart <==
1: // Regression test for a bad assertion in dart2js (can't compute subclasses of
2: // C because C isn't recorded as instantiated, which it really is, it's just
3: // that a compile-time error was encountered when attempting to resolve C).
4: class C {
5:   int sync*;
6: }
7: main() {
8:   new C();
9: }

org.dartlang.fletch.test-case:/main.dart:5:3:
Error: Can't have modifier 'int' here.
Try removing 'int'.
  int sync*;
  ^^^
Added breakpoint: id: '0' method: 'main' bytecode index: '0'
fletch_vm_stdout: Compile error
### 'finish' cancelled because another breakpoint was hit
Stack trace:
> 0: yield        	
  1: compileError 	@yield(InterruptKind.compileTimeError.index)
  2: C initializer	
  3: main         	@new C()
  4: callMain     	@invokeMain(arguments)
  5: entry        	@callMain([] )

Stack trace:
  0: yield        	
  1: compileError 	@yield(InterruptKind.compileTimeError.index)
  2: C initializer	
  3: main         	@new C()
> 4: callMain     	@invokeMain(arguments)
  5: entry        	@callMain([] )

Compile-time error expected
Program version 2 #10:
==> main.dart <==
1: // Regression test for a bad assertion in dart2js (can't compute subclasses of
2: // C because C isn't recorded as instantiated, which it really is, it's just
3: // that a compile-time error was encountered when attempting to resolve C).
4: class C {
5:   // TODO(ahe): There's no compile-time error here
6:   int sync;
7: }
8: main() {
9:   new C();
10: }

Reusing origin library(dart:core) (assumed read-only).
Reusing library(dart:_fletch_system) (assumed read-only).
Reusing origin library(dart:_internal) (assumed read-only).
Reusing origin library(dart:collection) (assumed read-only).
Reusing origin library(dart:math) (assumed read-only).
Reusing library(dart:fletch) (assumed read-only).
Reusing origin library(dart:convert) (assumed read-only).
Reusing origin library(dart:async) (assumed read-only).
Reusing library(dart:fletch.os) (assumed read-only).
Reusing library(dart:fletch.ffi) (assumed read-only).
Reusing origin library(dart:typed_data) (assumed read-only).
Attempting to reuse library(org.dartlang.fletch.test-case:/main.dart).
New library synthesized.
Differences computed.
Looking at difference: Modified(class(C) -> class(C))
Simple modification of class(C) detected
Differences computed.
Looking at difference: Added(Instance of 'PartialFieldList')
Add field field(C#sync) to class(C).
CommandCode.PrepareForChanges()
CommandCode.PushFromMap(MapId.classes, 0)
CommandCode.PushNewInteger(17409)
CommandCode.PushFromMap(MapId.methods, 265)
CommandCode.PushNewInteger(34817)
CommandCode.PushFromMap(MapId.methods, 17)
CommandCode.PushNewInteger(35840)
CommandCode.PushFromMap(MapId.methods, 16)
CommandCode.PushNewInteger(56321)
CommandCode.PushFromMap(MapId.methods, 18)
CommandCode.PushNewInteger(66560)
CommandCode.PushFromMap(MapId.methods, 346)
CommandCode.PushNewInteger(163072)
CommandCode.PushFromMap(MapId.methods, 0)
CommandCode.ChangeMethodTable(6)
CommandCode.PushFromMap(MapId.classes, 25)
CommandCode.PushNewInteger(194816)
CommandCode.PushFromMap(MapId.methods, 0)
CommandCode.PushNewInteger(216320)
CommandCode.PushFromMap(MapId.methods, 0)
CommandCode.PushNewInteger(216577)
CommandCode.PushFromMap(MapId.methods, 2)
CommandCode.ChangeMethodTable(3)
CommandCode.PushFromMap(MapId.classes, 25)
CommandCode.PushFromMap(MapId.classes, 0)
CommandCode.ChangeSuperClass()
CommandCode.PushFromMap(MapId.classes, 25)
CommandCode.PushNewInteger(0)
CommandCode.PushNull()
CommandCode.PushNewArray(2)
CommandCode.ChangeSchemas(1, 1)
CommandCode.CommitChanges(4)
Added breakpoint: id: '0' method: 'main' bytecode index: '0'
Stack trace:
> 0: main    	@new C()
  1: callMain	@invokeMain(arguments)
  2: entry   	@callMain([] )

fletch_vm_stdout: Compile error
### 'finish' cancelled because another breakpoint was hit
Stack trace:
> 0: yield        	
  1: compileError 	@yield(InterruptKind.compileTimeError.index)
  2: C initializer	
  3: main         	@new C()
  4: callMain     	@invokeMain(arguments)
  5: entry        	@callMain([] )

Stack trace:
  0: yield        	
  1: compileError 	@yield(InterruptKind.compileTimeError.index)
  2: C initializer	
  3: main         	@new C()
> 4: callMain     	@invokeMain(arguments)
  5: entry        	@callMain([] )

Compile-time error expected

Command[fletch_test]: FletchTestCommand(incremental/fix_compile_time_error_in_field)
Took 0:01:00.040139

Short reproduction command (experimental):
    python tools/test.py --kill-persistent-process=0 --run-gclient-hooks=0 --build-before-testing=0 --clang -rfletch_tests -t60 fletch_tests/incremental/fix_compile_time_error_in_field

```

peter-ahe-google, 2015-09-30:
This happened when building 12025edc53b09898372ffd4ea08baa2c1084ac5c.

peter-ahe-google, 2015-09-30:
Just saw this locally on `none-fletch_tests debug_ia32_clang fletch_tests/incremental/compile_time_error_002`. I ran lldb and saw:

```
(lldb) process attach --pid 55033
Process 55033 stopped
* thread #1: tid = 0xbe9a93, 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10
libsystem_kernel.dylib`__psynch_cvwait:
->  0x90d6a512 <+10>: jae    0x90d6a522                ; <+26>
    0x90d6a514 <+12>: calll  0x90d6a519                ; <+17>
    0x90d6a519 <+17>: popl   %edx
    0x90d6a51a <+18>: movl   0xf811b07(%edx), %edx

Executable module set to "/Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/fletch-vm".
Architecture set to: i386-apple-macosx.
(lldb) bt
* thread #1: tid = 0xbe9a93, 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  * frame #0: 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #1: 0x90ca4163 libsystem_pthread.dylib`_pthread_cond_wait + 726
    frame #2: 0x90ca7528 libsystem_pthread.dylib`pthread_cond_wait$UNIX2003 + 71
    frame #3: 0x00031e08 fletch-vm`fletch::MonitorImpl::Wait(this=0x023004f0) + 24 at platform_posix.h:54
    frame #4: 0x00025e6b fletch-vm`fletch::Scheduler::StopProgram(this=<unavailable>, program=<unavailable>) + 219 at scheduler.cc:109
    frame #5: 0x00027344 fletch-vm`fletch::Scheduler::Run(this=<unavailable>) + 452 at scheduler.cc:286
    frame #6: 0x0002f5f8 fletch-vm`fletch::Session::ProcessRun(this=0x0002718e) + 200 at session.cc:686
    frame #7: 0x000022ef fletch-vm`fletch::RunSession(connection=<unavailable>) + 47 at main.cc:27
    frame #8: 0x00002044 fletch-vm`fletch::Main(argc=<unavailable>, argv=<unavailable>) + 1108 at main.cc:159
    frame #9: 0x00001be1 fletch-vm`main(argc=2, argv=0xbffffacc) + 17 at main.cc:171
    frame #10: 0x985406d9 libdyld.dylib`start + 1
(lldb) thread list
Process 55033 stopped
* thread #1: tid = 0xbe9a93, 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  thread #2: tid = 0xbe9b5c, 0x90d6a54a libsystem_kernel.dylib`__psynch_mutexwait + 10
  thread #3: tid = 0xbe9b5d, 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10
(lldb) thread select 2
* thread #2: tid = 0xbe9b5c, 0x90d6a54a libsystem_kernel.dylib`__psynch_mutexwait + 10
    frame #0: 0x90d6a54a libsystem_kernel.dylib`__psynch_mutexwait + 10
libsystem_kernel.dylib`__psynch_mutexwait:
->  0x90d6a54a <+10>: jae    0x90d6a55a                ; <+26>
    0x90d6a54c <+12>: calll  0x90d6a551                ; <+17>
    0x90d6a551 <+17>: popl   %edx
    0x90d6a552 <+18>: movl   0xf811acf(%edx), %edx
(lldb) bt
* thread #2: tid = 0xbe9b5c, 0x90d6a54a libsystem_kernel.dylib`__psynch_mutexwait + 10
  * frame #0: 0x90d6a54a libsystem_kernel.dylib`__psynch_mutexwait + 10
    frame #1: 0x90ca1130 libsystem_pthread.dylib`_pthread_mutex_lock + 520
    frame #2: 0x90ca0f23 libsystem_pthread.dylib`pthread_mutex_lock + 16
    frame #3: 0x00027cb2 fletch-vm`fletch::Scheduler::RunInThread(this=0xbffff4e0) + 466 at scheduler.cc:488
    frame #4: 0x00037861 fletch-vm`fletch::ThreadPool::RunThread(arg=0x02300640) + 17 at thread_pool.cc:82
    frame #5: 0x90ca3c25 libsystem_pthread.dylib`_pthread_body + 138
    frame #6: 0x90ca3b9b libsystem_pthread.dylib`_pthread_start + 162
    frame #7: 0x90ca0e32 libsystem_pthread.dylib`thread_start + 34
(lldb) thread select 3
* thread #3: tid = 0xbe9b5d, 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #0: 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10
libsystem_kernel.dylib`__psynch_cvwait:
->  0x90d6a512 <+10>: jae    0x90d6a522                ; <+26>
    0x90d6a514 <+12>: calll  0x90d6a519                ; <+17>
    0x90d6a519 <+17>: popl   %edx
    0x90d6a51a <+18>: movl   0xf811b07(%edx), %edx
(lldb) bt
* thread #3: tid = 0xbe9b5d, 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10
  * frame #0: 0x90d6a512 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #1: 0x90ca4163 libsystem_pthread.dylib`_pthread_cond_wait + 726
    frame #2: 0x90ca7528 libsystem_pthread.dylib`pthread_cond_wait$UNIX2003 + 71
    frame #3: 0x00031e08 fletch-vm`fletch::MonitorImpl::Wait(this=0x023005a0) + 24 at platform_posix.h:54
    frame #4: 0x0000815b fletch-vm`fletch::GCThread::MainLoop(this=<unavailable>) + 59 at gc_thread.cc:121
    frame #5: 0x00008111 fletch-vm`fletch::GCThread::GCThreadEntryPoint(data=0x023004c0) + 17 at gc_thread.cc:14
    frame #6: 0x90ca3c25 libsystem_pthread.dylib`_pthread_body + 138
    frame #7: 0x90ca3b9b libsystem_pthread.dylib`_pthread_start + 162
    frame #8: 0x90ca0e32 libsystem_pthread.dylib`thread_start + 34
(lldb) 
```

peter-ahe-google, 2015-09-30:
This is not just about incremental tests. It can also affect non-incremental tests:

```
FAILED: none-fletchc debug_ia32_clang language/black_listed_test/12
Expected: Pass 
Actual: Crash
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugIA32Clang/fletch-vm &
$ out/DebugIA32Clang/fletch attach tcp_socket 127.0.0.1:60987 in session 7

 => 0
$ out/DebugIA32Clang/fletch run /mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/generated_tests/language/black_listed_test_12.dart in session 7
/mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/generated_tests/language/black_listed_test_12.dart:23:56:
Error: Unexpected token 'default'.
abstract class MyFunctionInterface implements Function default F {  /// 12: compile-time error
                                                       ^^^^^^^
/mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/generated_tests/language/black_listed_test_12.dart:
Warning: Could not find 'main'.
Try adding a method named 'main' to your program.

Compile error
[7: 2015-09-30 10:26:48.351 created session with file:///mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/.fletch-settings Settings(packages: file:///mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/.packages, options: [], constants: {}, device_address: null)]
[7: 2015-09-30 10:26:48.352 Connected to TCP vmSocket  vmSocket 46276 -> 127.0.0.1:60987]
[7: 2015-09-30 10:26:48.920 Compiled 'file:///mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/generated_tests/language/black_listed_test_12.dart' to 3509 commands]
[7: 2015-09-30 10:26:49.004 Compile-time error]

Timed out waiting for Fletch VM to shutdown; killing session
RangeError (index): Invalid value: Not in range 0..4, inclusive: -1
#0      List.[] (dart:core-patch/array.dart:11)
#1      StackTrace.visibleFrame (package:fletchc/stack_trace.dart:153:23)
#2      StackTrace.list (package:fletchc/stack_trace.dart:161:12)
#3      DebugState.list (package:fletchc/debug_state.dart:107:30)
#4      Session.list.<list_async_body> (package:fletchc/session.dart:542:23)
#5      _rootRunUnary (dart:async/zone.dart:911)
#6      _CustomZone.runUnary (dart:async/zone.dart:811)
#7      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#8      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#9      _Future._completeWithValue (dart:async/future_impl.dart:368)
#10     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#11     _rootRun (dart:async/zone.dart:904)
#12     _CustomZone.run (dart:async/zone.dart:803)
#13     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#14     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#15     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#16     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#17     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

 => 253
$ out/DebugIA32Clang/fletch x-end session 7
Ended session '7'.

 => 0
$ out/DebugIA32Clang/fletch-vm
Compile error

stderr:
Fletch VM exitcode is -15

Command[fletch_session]: /mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/tools/testing/bin/linux/dart -c -Dtest.fletch_session_command.is_incremental_enabled=false tools/testing/dart/fletch_session_command.dart out/DebugIA32Clang/fletch /mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/generated_tests/language/black_listed_test_12.dart
OR
gdb -ex 'follow-fork-mode child' -ex run --args out/DebugIA32Clang/dart -c -ppackage/ tests/fletchc/run.dart /mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/generated_tests/language/black_listed_test_12.dart
OR
In one terminal:
  gdb -ex run --args out/DebugIA32Clang/fletch-vm --port=54321
In another terminal:
  out/DebugIA32Clang/dart -c -ppackage/ -DattachToVm=54321 tests/fletchc/run.dart /mnt/data/b/build/slave/fletch-linux-debug-x86/build/fletch/out/DebugIA32Clang/generated_tests/language/black_listed_test_12.dart
Took 0:02:00.006734

Short reproduction command (experimental):
    python tools/test.py --kill-persistent-process=0 --run-gclient-hooks=0 --build-before-testing=0 --clang -rfletchc -t120 language/black_listed_test/12
```

peter-ahe-google, 2015-09-30:
And here's one with an ASAN failure:

```
FAILED: none-fletchc release_ia32_asan corelib/expando_test
Expected: CompileTimeError 
Actual: Crash
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/ReleaseIA32Asan/fletch-vm &
$ out/ReleaseIA32Asan/fletch attach tcp_socket 127.0.0.1:41806 in session 3

 => 0
$ out/ReleaseIA32Asan/fletch run /mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/tests/corelib/expando_test.dart in session 3
/mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/core/expando.dart:39:3:
Error: External function is not supported
  external Expando([String name]);
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
/mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/core/expando.dart:53:3:
Error: External function is not supported
  external T operator [](Object object);
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
/mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/core/expando.dart:62:3:
Error: External function is not supported
  external void operator []=(Object object, T value);
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Compile error
[3: 2015-09-30 08:39:16.519 created session with file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/.fletch-settings Settings(packages: file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/.packages, options: [], constants: {}, device_address: null)]
[3: 2015-09-30 08:39:16.519 Connected to TCP vmSocket  vmSocket 42955 -> 127.0.0.1:41806]
[3: 2015-09-30 08:39:17.079 Compiled 'file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/tests/corelib/expando_test.dart' to 5352 commands]
[3: 2015-09-30 08:39:17.276 Compile-time error]
'package:fletchc/src/fletch_backend.dart': Failed assertion: line 711 pos 12: 'Bytecode.identicalBytecodes(expectedBytecodes,
                                       codegen.assembler.bytecodes)' is not true.
#0      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:27)
#1      FletchBackend.createDebugInfo (package:fletchc/src/fletch_backend.dart:711:12)
#2      IncrementalCompiler.createDebugInfo (package:fletchc/incremental/fletchc_incremental.dart:200:38)
#3      DebugState.getDebugInfo.<anonymous closure> (package:fletchc/debug_state.dart:76:31)
#4      _HashVMBase&MapMixin&&_LinkedHashMapMixin.putIfAbsent (dart:collection-patch/compact_hash.dart:254)
#5      DebugState.getDebugInfo (package:fletchc/debug_state.dart:75:23)
#6      StackFrame.debugInfo (package:fletchc/stack_trace.dart:31:41)
#7      StackFrame.list (package:fletchc/stack_trace.dart:34:12)
#8      StackTrace.list (package:fletchc/stack_trace.dart:161:32)
#9      DebugState.list (package:fletchc/debug_state.dart:107:30)
#10     Session.list.<list_async_body> (package:fletchc/session.dart:542:23)
#11     _rootRunUnary (dart:async/zone.dart:911)
#12     _CustomZone.runUnary (dart:async/zone.dart:811)
#13     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#14     _Future._propagateToListeners (dart:async/future_impl.dart:577)
#15     _Future._completeWithValue (dart:async/future_impl.dart:368)
#16     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#17     _rootRun (dart:async/zone.dart:904)
#18     _CustomZone.run (dart:async/zone.dart:803)
#19     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#20     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#21     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#22     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#23     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

 => 253
$ out/ReleaseIA32Asan/fletch x-end session 3
Ended session '3'.

 => 0
$ out/ReleaseIA32Asan/fletch-vm
Compile error
$ out/ReleaseIA32Asan/fletch-vm
=================================================================
==13395== ERROR: AddressSanitizer: heap-use-after-free on address 0xf4205db8 at pc 0x80be218 bp 0xf4b7bf78 sp 0xf4b7bf6c
[... 43 lines cut ...]
  Stack after return:    f5
  Stack use after scope: f8
  Global redzone:        f9
  Global init order:     f6
  Poisoned by user:      f7
  ASan internal:         fe
==13395== ABORTING

stderr:
Fletch VM exitcode is -6
Non-zero exit code (-6) from: out/ReleaseIA32Asan/fletch-vm 

Command[fletch_session]: /mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/tools/testing/bin/linux/dart -c -Dtest.fletch_session_command.is_incremental_enabled=false tools/testing/dart/fletch_session_command.dart out/ReleaseIA32Asan/fletch /mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/tests/corelib/expando_test.dart
OR
gdb -ex 'follow-fork-mode child' -ex run --args out/ReleaseIA32Asan/dart -c -ppackage/ tests/fletchc/run.dart /mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/tests/corelib/expando_test.dart
OR
In one terminal:
  gdb -ex run --args out/ReleaseIA32Asan/fletch-vm --port=54321
In another terminal:
  out/ReleaseIA32Asan/dart -c -ppackage/ -DattachToVm=54321 tests/fletchc/run.dart /mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/tests/corelib/expando_test.dart
Took 0:00:01.077211

Short reproduction command (experimental):
    python tools/test.py -mrelease --kill-persistent-process=0 --run-gclient-hooks=0 --build-before-testing=0 --asan -rfletchc -t60 corelib/expando_test
```

peter-ahe-google, 2015-09-30:
I just had this test failing locally:

```
FAILED: fletchc-fletchvm debug_ia32_clang corelib/int_modulo_arith_test/modPow
Expected: RuntimeError 
Actual: Timeout
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugIA32Clang/fletch-vm &
$ out/DebugIA32Clang/fletch attach tcp_socket 127.0.0.1:49212 in session 6

 => 0
$ out/DebugIA32Clang/fletch export /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_tests/corelib/int_modulo_arith_test_modPow.dart to file /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_corelib_int_modulo_arith_test_modPow/fletch.snapshot in session 6

 => 0
$ out/DebugIA32Clang/fletch x-end session 6
Ended session '6'.

 => 0
$ out/DebugIA32Clang/fletch-vm

stderr:
Fletch VM exitcode is 0

CommandOutput[vm]:

stdout:
Uncaught exception:
a Instance: 
  - class = class  - @0 = null
  - @1 = <unknown>
  - @2 = null
  - @3 = null
  - @4 = null

Command[fletch_session]: /Users/ahe/Dart/fletch-repo/fletch/tools/testing/bin/mac/dart -c -Dtest.fletch_session_command.is_incremental_enabled=false tools/testing/dart/fletch_session_command.dart out/DebugIA32Clang/fletch /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_tests/corelib/int_modulo_arith_test_modPow.dart
OR
gdb -ex 'follow-fork-mode child' -ex run --args out/DebugIA32Clang/dart -c -ppackage/ tests/fletchc/run.dart /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_tests/corelib/int_modulo_arith_test_modPow.dart
OR
In one terminal:
  gdb -ex run --args out/DebugIA32Clang/fletch-vm --port=54321
In another terminal:
  out/DebugIA32Clang/dart -c -ppackage/ -DattachToVm=54321 tests/fletchc/run.dart /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_tests/corelib/int_modulo_arith_test_modPow.dart
Took 0:00:00.853911

Command[vm]: out/DebugIA32Clang/fletch-vm /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_corelib_int_modulo_arith_test_modPow/fletch.snapshot
Took 0:02:00.011000

Command[vm]: out/DebugIA32Clang/fletch-vm -Xunfold-program /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_corelib_int_modulo_arith_test_modPow/fletch.snapshot
Did not run

Short reproduction command (experimental):
    python tools/test.py -rfletchvm -cfletchc -t120 corelib/int_modulo_arith_test/modPow
```

Notice that the command that timed out was:

```out/DebugIA32Clang/fletch-vm /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_corelib_int_modulo_arith_test_modPow/fletch.snapshot```

When I rerun that, it completes in less than 300ms.

mkustermann, 2015-10-01:
Proposed CL for fixing the deadlock: https://codereview.chromium.org/1376123003/

peter-ahe-google, 2015-10-02:
Fixed in 8149fff36ff843b623421cbd5a73e7c05cc5e3e5

closed 2015-10-02 by mkustermann