dart-archive/sdk #163: Request: Support PWM over GPIO

mit-mit, 2015-09-30:
Would like the ability to set a pin to an 'analog' value rather than just on/off. See:
https://learn.adafruit.com/adafruit-arduino-lesson-3-rgb-leds/theory-pwm

marloncarvalho, 2016-07-16:
Is there any plan to release this feature? 

I'm trying to control a continuous rotation servo motor using Dartino but I think it's only possible using PWM (at least, my current Python code does it using PWM). Do you have any other suggestion? Are there other ways to control this kind of servo motor using Dartino?

mit-mit, 2016-07-18:
Yes, we are still considering this, but have no concrete timeline. Are you using an STM board, or a Pi?

marloncarvalho, 2016-07-18:
Hi @mit-mit, I'm using a Pi 2. I'm working on a project that has a servo, HC SR04, and a LCD 16x2. I'd like to use Dartino to control all those devices. I've been using Dart for web development and it would be really great if I could use it in my IoT projects too! :-)