dart-archive/sdk #164: [SDK] Clean up bin/

kasperl, 2015-10-01:
How many of the files in bin/ are useful to end-users? 

```
  bin/fletch-sdk.packages
  bin/fletch
  bin/fletch-vm
  bin/natives.json
  bin/fletch-agent.snapshot
```

Can we get it down to one or two? Clearly we need `fletch`, but I guess the only reason why we have `fletch-vm` in there is if our users want to start from a snapshot? Maybe it would be easier to grok if it was called `fletch-run-snapshot` in that context?

peter-ahe-google, 2015-10-01:
It should be fine to have only `fletch` and `fletch-vm` in `bin`. I see the attraction of renaming `fletch-vm`, but I also think it serves a purpose as a standalone program you can explicitly connect to. So my inclination is to keep the name.

kasperl, 2015-10-01:
I think we'll find lots of people who will try to run `fletch-vm hello.dart` as their first program. At least I know of more than one who has already tried that.

peter-ahe-google, 2015-10-01:
Perhaps we should fix that by producing a sensible error message?

sgjesse, 2015-10-01:
The `bin/fletch-agent.snapshot` file is now removed.

sgjesse, 2015-10-01:
As for `bin/natives.json` is that used by the compiler? If so where should it be?

mit-mit, 2015-10-02:
Min bar for launch is at least splitting into two:

Keep in bin:
```
bin/fletch
bin/fletch-vm
```

Move to something internal.
```
bin/natives.json
```

commit 474c243b (2015-10-02): Look for natives.json relative to the dart executable, not the vm.

R=kasperl@google.com, ahe@google.com

Closes #164

Review URL: https://codereview.chromium.org/1380303002 .
closed 2015-10-02 by ricowind