dart-archive/sdk #168: Create mock versions of gpio and os packages

mit-mit, 2015-10-01:
Needed os that we can illustrate running the fletch program on the developer PC.

peter-ahe-google, 2015-10-04:
We need a mock packages configuration file in an internal directory of the SDK. In addition, we need to change the local.fletch-settings file that is generated in [developer.dart](https://github.com/dart-lang/fletch/blob/ffc031ca80fbe339afea70ba51b9acbd4c5ff7aa/pkg/fletchc/lib/src/driver/developer.dart#L268).

mit-mit, 2015-10-05:
Søren has a mock of gpio here:
https://codereview.chromium.org/1389573002/

sgjesse, 2015-10-30:
GPIO mock landed in https://github.com/dart-lang/fletch/commit/2ca94db0117a85c981a8042b6ae735e4fd578e53.

I am not sure making a drop-in mock replacement of GIPO in a separate package is feasible.

closed 2015-10-30 by sgjesse