dart-archive/sdk #172: fletch reset

mit-mit, 2015-10-01:
If something hangs, you have to ```killall dart```. Would be good to have something like ```fletch reset``` or ```fletch abort``` instead.

peter-ahe-google, 2015-10-01:
@lukechurch Suggestions for verb name?

peter-ahe-google, 2015-10-01:
I like `fletch force quit`, but I'm a Mac user. Perhaps `fletch quit` or `fletch exit` for Mac and Windows users respectively.

lukechurch, 2015-10-02:
As discussed
```
fletch force quit
```
To terminate everything, whether or not it is 'responding'. In the future 
```
fletch shutdown all
```
Will politely request all of the clients to close.

peter-ahe-google, 2015-10-03:
After further considerations, we realized that `fletch quit` would be fine. Rather than having two quit commands (one for `kill` and one for `kill -9`) we realized that we just need one command that will first attempt `kill`. If the persistent process hasn't exited after 2 seconds, the command will use `kill -9`.

peter-ahe-google, 2015-10-03:
Proposed fix in [CL 1374043006](https://codereview.chromium.org/1374043006/)

commit b9ea7910 (2015-10-04): Add 'fletch quit' command

Fixes #172

R=lukechurch@google.com, wibling@google.com

Review URL: https://codereview.chromium.org/1374043006 .
closed 2015-10-04 by peter-ahe-google