dart-archive/sdk #176: Reduce the set of libraries shipped in the Fletch SDK

mit-mit, 2015-10-02:


mit-mit, 2015-10-05:
Moving to 0.1.1

sigurdm, 2015-12-22:
With e21339a0f7b7ae9d545cb9cdb50f4b93873b432b the libraries available when using "device-type: embedded" are:

fletch.ffi: ffi/ffi.dart
fletch: fletch/fletch.dart
fletch.io: io/io.dart
fletch.service: service/service.dart
fletch.os: os/os.dart
fletch._system: system/system.dart

# Embedded core is a reduced version of the regular dart:core.
core: core/embedded_core.dart

math: ../third_party/dart/sdk/lib/math/math.dart
typed_data: ../third_party/dart/sdk/lib/typed_data/typed_data.dart
developer: ../third_party/dart/sdk/lib/developer/developer.dart
collection: ../third_party/dart/sdk/lib/collection/collection.dart

Still needed is to create packages for Uri and Regexp.
And to decide if we really want to remove collection (core depends on collection).

closed 2015-12-22 by sigurdm