dart-archive/sdk #182: Making it easy to manually run debugger tests and update expectation files.

mkustermann, 2015-10-02:
The change 995b40aac63610d1b4e5f0707a9e88a319c34035 caused us to loose a way of quickly reproducing debugger test expectation mismatch.

Previously `./tools/test.py` provided a simple reproduction command, like this:
```
$ out/DebugIA32/fletch compile-and-run --test-debugger=r,q tests/debugger/unsupported_hint_test.dart
```
So one could easily update the expectation file.

Now the output of `./tools/test.py` is not useful anymore and it became much harder to produce a new expectation file:
```
  Command[fletch_test]: FletchTestCommand(debugger/unsupported_hint_test)
```

Maybe we can have a small shell script, or something like that, that uses `fletch debug` with the correct debugging commands to reproduce test failures?

cc: @peter-ahe-google

mkustermann, 2015-10-02:
What makes this much worse is that the debugger output uses tab characters (i.e. '\t' characters). Using copy&paste with mouse seems to make them spaces. This is really bad!

madsager, 2015-10-02:
#!/bin/bash

FLETCH_DIR=/home/ager/ssd/fletch-repo/fletch

cd $FLETCH_DIR

./out/ReleaseIA32/fletch shutdown;

for file in `ls tests/debugger/*_test.dart`;
do
    echo $file
    TEST_DEBUGGER=`grep "FletchDebuggerCommands=" $file`
    if [ -n "$TEST_DEBUGGER" ]; then
        TEST_DEBUGGER="${TEST_DEBUGGER/\/\/ FletchDebuggerCommands/}"
    fi
    EXPECTATION_FILE="${file/_test.dart/_expected.txt}";
    ./out/ReleaseIA32/fletch compile-and-run $file "--test-debugger$TEST_DEBUGGER" > $EXPECTATION_FILE;
done

madsager, 2015-10-02:
I have used the above script to rebaseline debugger tests in the past. It was something I put together very quickly and I don't know if it still works so YMMV.

Feel free to change the debugger to not use tabs.

madsager, 2015-10-02:
Oh, I see, we removed compile-and-run and I haven't been following the changes in connection with that. So I don't know what the right replacement is in the script above.

mkustermann, 2015-10-07:
I'm running into this issue quite often and am impatient. I'll take it over.

mkustermann, 2015-11-23:
This has been fixed for a while. One should be able to generate new golden files via:
```
$ ./out/DebugIA32/dart -Dfletch.version=`out/DebugIA32/fletch-vm --version`  tests/debugger/debugger_tests.dart             
```

closed 2015-11-23 by mkustermann