dart-archive/sdk #185: Flaky vm crash on the sdk builder on unsorted/1_test

ricowind, 2015-10-03:
Snippet:
$ out/ReleaseX64Clang/fletch-sdk/bin/fletch-vm
6
$ out/ReleaseX64Clang/fletch-sdk/bin/fletch-vm
fletch-vm(43332,0x103a91000) malloc: *** error for object 0x7fc834809e00: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug

stderr:
Fletch VM exitcode is -6
Non-zero exit code (-6) from: out/ReleaseX64Clang/fletch-sdk/bin/fletch-vm 

Full log:
https://uberchromegw.corp.google.com/i/client.dart.internal/builders/fletch-mac-release-x86-sdk/builds/19/steps/test_ReleaseX64Clang%20failures/logs/stdio

cc @madsager @mkustermann

mkustermann, 2015-10-03:
That is a pretty weired test. It will always fail after just executing a few bytecodes (i.e. pretty fast), since it treats the `List<String> arguments` to `main()` as an `int` and tries to add a number to it.

The information in the buildbot failure is not enough to debug it and I can't reproduce it.

Also a double free (or a free of a non allocated address) should be caught by our asan builders, where we can get much better information.

[Sidenote: The failure happened not on a dev branch commit but on a master commit -- seems like someone triggered a build on sdk builders with a master commit. ]

ricowind, 2015-10-03:
Sidenote on your sidenote: this was not the dev branch sdk builder but the master branch sdk builder. You are correct that it was a triggered build though - by me

mkustermann, 2015-10-08:
Another (related or unrelated) failure on mac is here:
https://uberchromegw.corp.google.com/i/client.dart.internal/builders/fletch-mac-release-x86/builds/833

This time we have a coredump, so we might be able to take a look at where it crashed.

peter-ahe-google, 2015-10-08:
I've extracted this information:

```
(lldb) target create "fletch-vm" --core "core.2376"
Core file '/Users/ahe/Dart/fletch-repo/fletch/core.2376' (i386) was loaded.
Process 0 stopped
* thread #1: tid = 0x0000, 0x914d9802 libsystem_kernel.dylib`__psynch_mutexwait + 10, stop reason = signal SIGSTOP
    frame #0: 0x914d9802 libsystem_kernel.dylib`__psynch_mutexwait + 10
libsystem_kernel.dylib`__psynch_mutexwait:
->  0x914d9802 <+10>: jae    0x914d9812                ; <+26>
    0x914d9804 <+12>: calll  0x914d9809                ; <+17>
    0x914d9809 <+17>: popl   %edx
    0x914d980a <+18>: movl   0xeeaa817(%edx), %edx
  thread #2: tid = 0x0001, 0x914d9952 libsystem_kernel.dylib`__pthread_kill + 10, stop reason = signal SIGSTOP
    frame #0: 0x914d9952 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x914d9952 <+10>: jae    0x914d9962                ; <+26>
    0x914d9954 <+12>: calll  0x914d9959                ; <+17>
    0x914d9959 <+17>: popl   %edx
    0x914d995a <+18>: movl   0xeeaa6c7(%edx), %edx
(lldb) thread list
Process 0 stopped
* thread #1: tid = 0x0000, 0x914d9802 libsystem_kernel.dylib`__psynch_mutexwait + 10, stop reason = signal SIGSTOP
  thread #2: tid = 0x0001, 0x914d9952 libsystem_kernel.dylib`__pthread_kill + 10, stop reason = signal SIGSTOP
(lldb) bt
* thread #1: tid = 0x0000, 0x914d9802 libsystem_kernel.dylib`__psynch_mutexwait + 10, stop reason = signal SIGSTOP
  * frame #0: 0x914d9802 libsystem_kernel.dylib`__psynch_mutexwait + 10
    frame #1: 0x9bd93945 libsystem_pthread.dylib`_pthread_mutex_lock + 404
    frame #2: 0x9bd937ac libsystem_pthread.dylib`pthread_mutex_lock + 16
    frame #3: 0x0001f4b7 fletch-vm`fletch::Session::ProcessRun() + 279
    frame #4: 0x00001ae5 fletch-vm`main + 1077
    frame #5: 0x961e9701 libdyld.dylib`start + 1
(lldb) thread select 2
* thread #2: tid = 0x0001, 0x914d9952 libsystem_kernel.dylib`__pthread_kill + 10, stop reason = signal SIGSTOP
    frame #0: 0x914d9952 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x914d9952 <+10>: jae    0x914d9962                ; <+26>
    0x914d9954 <+12>: calll  0x914d9959                ; <+17>
    0x914d9959 <+17>: popl   %edx
    0x914d995a <+18>: movl   0xeeaa6c7(%edx), %edx
(lldb) bt
* thread #2: tid = 0x0001, 0x914d9952 libsystem_kernel.dylib`__pthread_kill + 10, stop reason = signal SIGSTOP
  * frame #0: 0x914d9952 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x9bd91167 libsystem_pthread.dylib`pthread_kill + 101
    frame #2: 0x9c1c529c libsystem_c.dylib`abort + 155
    frame #3: 0x9c20bdab libsystem_malloc.dylib`free + 419
    frame #4: 0x000225ba fletch-vm`fletch::StoreBuffer::~StoreBuffer() + 26
    frame #5: 0x00014837 fletch-vm`fletch::Process::~Process() + 119
    frame #6: 0x00016851 fletch-vm`fletch::Program::DeleteProcess(fletch::Process*) + 97
    frame #7: 0x0001b053 fletch-vm`fletch::Scheduler::ExitAtTermination(fletch::Process*) + 35
    frame #8: 0x0001e3b7 fletch-vm`fletch::Session::ProcessMessages() + 4999
    frame #9: 0x0001cabf fletch-vm`fletch::MessageProcessingThread(void*) + 15
    frame #10: 0x9bd905fb libsystem_pthread.dylib`_pthread_body + 144
    frame #11: 0x9bd90485 libsystem_pthread.dylib`_pthread_start + 130
    frame #12: 0x9bd95cf2 libsystem_pthread.dylib`thread_start + 34
(lldb) 
```

peter-ahe-google, 2015-10-08:
@ErikCorryGoogle

ErikCorryGoogle, 2015-10-08:
Peter and I took a look at the core file in lldb.

The process destructor calls the store buffer destructor because of the part-object relationship.
Freeing the very first chunk in the store buffer triggers the already-freed error from the malloc library.
Our best guess is that two threads are deleting the same process at once, since there is no obvious lock preventing this.  There is locking when the process is removed from the process linked-list, but it is not this removal that triggers the deletion of the process, and it looks like the linked-list removal fails fairly harmlessly if the process was already removed. We couldn't see why more than one thread would delete the process, but it seems likely that there's a way for this to happen.

In other words we didn't find the root cause.  It's a release binary, but I'm not sure there would be more information in a debug binary.  The other thread did not seem to be doing anything dangerous in the stack trace in the core file.

mkustermann, 2015-10-08:
Thank you two for getting this information. It provides enough detail to conclude where the race condition is. One way to fix this is here: https://codereview.chromium.org/1400573002/

mit-mit, 2016-05-30:
Closing as this landed on Oct 9th.

closed 2016-05-30 by mit-mit