dart-archive/sdk #187: Running on the raspberry through the agent fails

ricowind, 2015-10-04:
So I just tried out running the agent on my raspberry, it does not really work.
I installed the deb package
did "bin/fletch attach tcp_socket 192.168.1.143:12121"
did "bin/fletch run samples/general/hello.dart in session remote"

```
Bad state: Trying to send command CommandCode.NewMap(MapId.methods) to fletch-vm, but the connection is already closed.
#0      FletchVmSession.sendCommand.<sendCommand_async_body> (package:fletchc/session.dart:124:7)
#1      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#2      _rootRun (dart:async/zone.dart:904)
#3      _CustomZone.run (dart:async/zone.dart:803)
#4      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#5      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#6      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#7      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#8      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

On the pi, from the log
```
cat /var/log/fletch-agent.log 
2015-10-04 11:38:39.284 Agent INFO: Agent log file: /var/log/fletch-agent.log
2015-10-04 11:38:39.288 Agent INFO: Agent pid file: /var/run/fletch-agent.pid
2015-10-04 11:38:39.290 Agent INFO: Vm path: /usr/lib/fletch-agent/bin/fletch-vm
2015-10-04 11:38:39.292 Agent INFO: Log path: /var/log/fletch-agent
2015-10-04 11:38:39.294 Agent INFO: Run path: /var/run/fletch-agent
2015-10-04 11:38:39.298 Agent INFO: starting server on 0.0.0.0:12121
2015-10-04 11:44:52.013 Agent WARNING: Unknown command: 3072.
2015-10-04 11:47:43.364 Agent WARNING: Unknown command: 3072.
2015-10-04 11:50:10.494 Agent WARNING: Unknown command: 3072.
```

ricowind, 2015-10-04:
@wibling @sgjesse

kasperl, 2015-10-04:
I successfully used a device_address in a settings file without a port
number on Friday. I don't think you're meant to attach to the agent.

ricowind, 2015-10-04:
The agent is the only thing listening on a socket on the pi (well except my ssh deamon)

ricowind, 2015-10-04:
btw, this was on the sdk build, from d9d9dd5c7616ca235b1c8a4885c647250012d068

kasperl, 2015-10-04:
What I meant is you shouldn't use the 'attach' command. If you don't then fletch will ask you for an ip address when you first use session remote.

ricowind, 2015-10-04:
Nope, then it just runs it on my mac

sgjesse, 2015-10-04:
Kasper is right _attach_ is for attaching directly to an already started VM. You should use sessions with the agent, which will then start a new VM and attach each time you run.

If you build from master with the latest changes from Peter (https://github.com/dart-lang/fletch/commit/ffc031ca80fbe339afea70ba51b9acbd4c5ff7aa), you should be able to just run:

```
$ out/DebugX64Clang/fletch run samples/raspberry_pi/basic/knight-rider.dart in session remote
```

It will then ask for the IP of the remote if the file 'remote.fletch-settings' does not exist - you should see the following

```
Please enter IP address of remote device: pi3       
Created settings file 'remote.fletch-settings'
```

I entered _pi3_, which is the IP (in `/etc/hosts`) of the RPi I am currently using.

And the the program should run on the RPi.

Make sure to have a newly build agent from master as well, as I have seen compatibility issues with slightly older ones leading to segfault - maybe snapshot incompatibility - but I did not debug it.

Take a look at `remote.fletch-settings`, which now holds the IP of the remote.

You can also create you own my.fletch-settings with different settings, and start a session like this:

```
$ out/DebugX64Clang/fletch create session foo with my.fletch-settings
```

The you default session will be `foo` with the information from `my.fletch-settings`.

peter-ahe-google, 2015-10-04:
@kasperl and @sgjesse are correct. However, we should fix the attach command so that it either accepts an attempt to connect to the agent, or that it tells you what to do instead.

ricowind, 2015-10-04:
So I can confirm that it is working at TOT, sorry for the noise, I just took the latest sdk (which was 2 revisions old). I have forced a new sdk build now

wibling, 2015-10-06:
Changing the title according to @peter-ahe-google comment above.

closed 2016-05-30 by mit-mit