dart-archive/sdk #188: Give better error on input error not caught

ricowind, 2015-10-05:
Output from console on mac, started sample, unplugged network cable:
```
ricow-macbookpro:fletch-sdk ricow$ bin/fletch run samples/raspberry_pi/basic/blinky.dart in session remote
^C^C^C^C^C^C^C^C^C^C^CThe Fletch program is broken and has crashed.

The Fletch team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dart-lang/fletch/issues/new

Please include the following information:

* the name and version of your operating system

* the Fletch SDK version (0.1.0-edge.5015466f60b9f4a468500e346ead47efdd9e28d8)

* the entire message you see here (including the full stack trace below)

Uncaught Error: InputError(DiagnosticKind.socketAgentConnectError, {#{message}: Operation timed out, #{address}: 192.168.2.2:12121})
Stack Trace:
#0      throwFatalError (package:fletchc/src/diagnostic.dart:233:3)
#1      connect.<connect_async_body>.<anonymous closure> (package:fletchc/src/driver/developer.dart:101:9)
#2      _rootRunUnary (dart:async/zone.dart:911)
#3      _CustomZone.runUnary (dart:async/zone.dart:811)
#4      _Future._propagateToListeners.handleError (dart:async/future_impl.dart:525)
#5      _Future._propagateToListeners (dart:async/future_impl.dart:580)
#6      _Future._completeError (dart:async/future_impl.dart:376)
#7      _Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:431)
#8      _rootRun (dart:async/zone.dart:904)
#9      _CustomZone.run (dart:async/zone.dart:803)
#10     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#11     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#12     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#13     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#14     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

#0      _rootHandleUncaughtError.<anonymous closure> (dart:async/zone.dart:895)
#1      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#2      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#3      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#4      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

peter-ahe-google, 2015-10-07:
I've developed tests for this situation, see [CL 1389283002](https://codereview.chromium.org/1389283002/).

peter-ahe-google, 2015-10-07:
Proposed fix in [CL 1392893002](https://codereview.chromium.org/1392893002/).

commit 124d943f (2015-10-08): Improve handling of socket errors from VM connection

Fixes #188

R=wibling@google.com

Review URL: https://codereview.chromium.org/1392893002 .
closed 2015-10-08 by peter-ahe-google