dart-archive/sdk #189: The generated remote.fletch-settings have "null" in "packages"

sgjesse, 2015-10-05:
Using the [latest SDK](http://storage.googleapis.com/fletch-archive/channels/be/raw/0.1.0-edge.5015466f60b9f4a468500e346ead47efdd9e28d8/sdk/fletch-sdk-linux-ia32-release.zip), and running

```
$ bin/fletch run samples/raspberry_pi/basic/blinky.dart in session remote
```

Enter `pi3` when asked `Please enter IP address of remote device:`.

The created `remote.fletch-settings` have `"null"` (in quotes), and not just `null`:

```
{
  "packages": "null",
  "options": [],
  "constants": {},
  "device_address": "pi3:12121"
}
```

peter-ahe-google, 2015-10-05:
I cannot reproduce this problem.

peter-ahe-google, 2015-10-05:
I can repro, and have a fix.

peter-ahe-google, 2015-10-05:
Proposed fix: [CL 1384253005](https://codereview.chromium.org/1384253005/)

peter-ahe-google, 2015-10-05:
In order to reproduce this, I had to delete `fletch-sdk/internal/fletch-sdk.packages`.

closed 2015-10-05 by peter-ahe-google

sgjesse, 2015-10-05:
I don't know how I ended up in this situation. I had unpacked an SDK from the bot. I still had it in a temp directory. Removing `remote.fletch-settings` and running

```
$ bin/fletch run samples/raspberry_pi/basic/blinky.dart in session remote
```

again created this `remote.fletch-settings`

```
{
  "packages": "file:///tmp/fletch-sdk/fletch-sdk/internal/fletch-sdk.packages",
  "options": [],
  "constants": {},
  "device_address": "pi3:12121"
}
```

Running

```
$ ls -l internal/
```

Gives (with user and group renamed);

```
total 11504
-rwx------  1 uuu xxx 11740984 Oct  4 16:01 dart
drwx------  3 uuu xxx     4096 Oct  4 16:01 dart_lib
drwx------  3 uuu xxx     4096 Oct  4 16:01 fletch_lib
-rw-------  1 uuu xxx      161 Oct  4 16:01 fletch-sdk.packages
-rw-------  1 uuu xxx    18534 Oct  4 16:01 natives.json
drwx------ 13 uuu xxx     4096 Oct  4 16:01 pkg
```

so `internal/fletch-sdk.packages` seemed to have been there all along.

sgjesse, 2015-10-05:
Tried with a clean 0.1.0-dev.5.0. No problems. Created a `$HOME/remote.fletch-settings` which points to the correct `fletch-sdk.packages`.

commit aee9b4bf (2015-10-05): JSON serialize packages correctly when null

Fixes #189

R=wibling@google.com

Review URL: https://codereview.chromium.org/1384253005 .