dart-archive/sdk #191: hello.dart stack traces

lukechurch, 2015-10-05:
```
lukechurch-macbookpro:~ lukechurch$ cd ~/fletch-sdk/samples/general/
lukechurch-macbookpro:general lukechurch$ fletch run hello.dart 
/Users/lukechurch/fletch-sdk/samples/general/hello.dart:11:26:
Warning: No method named 'systemInformation' in class 'System'.
  SystemInformation si = sys.systemInformation();
                         ^^^^^^^^^^^^^^^^^^^^^^^
Uncaught exception: NoSuchMethodError: Class 'MacOSSystem' has no method named 'systemInformation'
Stack trace:
> 0: main                          	@sys.systemInformation()

/Users/lukechurch/fletch-sdk/samples/general/hello.dart:11:26
6    // it was run in.
7    library helloSample;
8    import 'package:os/os.dart';
9    
10   main() {
11     SystemInformation si = sys.systemInformation();
12     print('Hello from ${si.operatingSystemName} running on ${si.nodeName}.');
13   }
[default: 2015-10-05 09:34:19.748 created session]
[default: 2015-10-05 09:34:21.618 Compiled 'file:///Users/lukechurch/fletch-sdk/samples/general/hello.dart' to 3686 commands]
[default: 2015-10-05 09:34:21.666 Connected to TCP vmSocket  vmSocket 59578 -> 127.0.0.1:59577]
[default: 2015-10-05 09:34:21.901 Uncaught error]
lukechurch-macbookpro:general lukechurch$ 

```

lukechurch, 2015-10-05:
(0.1.0-dev.2.0 on OS X)

kasperl, 2015-10-05:
Appears fixed on bleeding edge.

lukechurch, 2015-10-05:
:-)

closed 2015-10-05 by lukechurch