dart-archive/sdk #192: Logging related to implicitly created session goes directly to stdout

kasperl, 2015-10-05:
Should this use the logging "infrastructure" instead so it doesn't get printed when everything just works, @peter-ahe-google?

```
kasperl$ bin/fletch run ../hello.dart 
Created settings file 'local.fletch-settings'
Hello, World
```

kasperl, 2015-10-05:
This also goes for the output from `fletch quit`.

```
kasperl$ bin/fletch quit
Background process exited
```

peter-ahe-google, 2015-10-05:
In both cases, I think the output is necessary. The files `local.fletch-settings` and `remote.fletch-settings` are created once. Especially for the `remote.fletch-settings` file, I think it is important for understanding why it doesn't ask you for an IP address ever again.

Similarly, the output from `fletch quit` reflects user feedback that it was hard to see if it worked or not.

peter-ahe-google, 2015-10-05:
@lukechurch WDYT?

kasperl, 2015-10-05:
Maybe the biggest issue that the messages should be more directed towards the user. I'll let @lukechurch try to figure out how to improve things, but in my opinion it's not clear why the user should care about the `remote.fletch-settings` file so maybe we should tell him/her?

Maybe the `fletch quit` should just report `Success.` if everything worked out?