dart-archive/sdk #193: Our documentation needs to work around issues decompressing Raspbian image

lukechurch, 2015-10-05:
The binaries from here: https://www.raspberrypi.org/downloads/raspbian/ don't decompress with the tools installed by default on OS X. Our documentation needs to explain the work arounds (e.g. User The Unarchvier)

Even though this doesn't feel like it's our problem the user impact is noticeable enough that we need to document the work around.

lukechurch, 2015-10-05:
The GUI tool decompresses it to a cpgz file. The command line tool reports
```
 "skipping: 2015-09-24-raspbian-jessie.img need PK compat. v4.5 (can do v2.1)"
```

lukechurch, 2015-10-05:
This only appears to impact the Jessie image, the Wheezy file seems to be ok.

ricowind, 2015-10-05:
You can do:
```
ditto -x -k 2015-09-24-raspbian-jessie.zip .
```
instead of using unzip, which is installed by default on newer OS X's

mit-mit, 2015-10-05:
Done, https://codereview.chromium.org/1383673003/

closed 2015-10-05 by mit-mit