dart-archive/sdk #194: Control-C leads to something that looks like errors

kasperl, 2015-10-05:
This is probably confusing to users. The user asked for stopping the VM (Control-C) and it worked, so there is probably no need to give them the "error" logs.

```
bin/fletch run samples/raspberry_pi/basic/blinky.dart in session remote
^C[remote: 2015-10-05 13:14:24.223 Compiled 'file:///Users/kasperl/Dart/fletch-sdk/samples/raspberry_pi/basic/blinky.dart' to 6509 commands]
[remote: 2015-10-05 13:14:24.231 Connected to TCP agentSocket  agentSocket 53670 -> 192.168.1.42:12121]
[remote: 2015-10-05 13:14:24.271 Connected to TCP vmSocket  vmSocket 53671 -> 192.168.1.42:51628]
[remote: 2015-10-05 13:14:29.464 Received signal 2]
[remote: 2015-10-05 13:14:29.466 Connected to TCP agentSocket  agentSocket 53672 -> 192.168.1.42:12121]
[remote: 2015-10-05 13:14:29.473 Error on connection to Fletch VM: Connection is closed]
```

kasperl, 2015-10-05:
@wibling Thoughts?

wibling, 2015-10-05:
Yes, we should try to filter these out. I have seen them, but not gotten
around to them yet.

wibling, 2015-10-05:
After a quick discussion with Peter we agreed to implement a "fletch show log" command which a user can use to get information about a session. This way we can avoid printing this information to stdout which is indeed misleading.
As a separate change we will also put the session into a state when exiting so the last line of the above log does not print as an error, since it is really expected behavior.

commit 936dc220 (2015-10-05): Create settings files in $HOME. Also, implement "fletch show log".

Fixes #196
Fixes #194

R=wibling@google.com

Review URL: https://codereview.chromium.org/1382043005 .
closed 2015-10-05 by peter-ahe-google