dart-archive/sdk #200: Analyzer support

mit-mit, 2015-10-05:
We should support the Analyzer to get a better developer experience. Right now editing projects, for example in Atom, causes lots of warnings such as ```Target of the URI does not exist: 'dart:fletch' ```

peter-ahe-google, 2015-10-05:
We're not going to get a better developer experience by supporting the Analyzer. The Analyzer needs to support us. This implies that analyzer front-end should be integrated with the fletch backend.

This work was started in September 2014, but little progress has been made, and I find it highly unrealistic to target this to 0.1.1.

mit-mit, 2015-10-05:
OK, I just tagged it to make sure we took time to discuss it :-)

kasperl, 2015-10-23:
We discussed this again a few days ago and it seems like this issue may no longer accurately reflect that plans for 0.1.1.

sgjesse, 2015-10-23:
We currently have the `_sdkext` in the fletch package, however with the smaller set of core libraries we might actually have to write a .analysis_options (https://codereview.chromium.org/1414133002/diff/40001/pkg/analyzer/README.md) to avoid getting wrong completion.

danrubel, 2016-01-07:
Per discussion with @sigurdm this morning, I opened [report an error on async usage in fletch projects](https://github.com/dart-lang/sdk/issues/25373).

mit-mit, 2016-02-18:
Required changes to the SDK landed in https://github.com/dartino/sdk/commit/5bdbefba6954180cd9984976e3a9c2fcb98b6140 and https://github.com/dartino/sdk/commit/9d91ba757dd3fb63f8a5b9629ab52799fc74f978.

Remaining changes will be in the atom package: https://github.com/dartino/atom-dartino/issues/13

closed 2016-02-18 by mit-mit