dart-archive/sdk #201: Help for debugger command 's' is not accurate

sgjesse, 2015-10-05:
Typing `help` at the debugger prompt show the following for the `b` command

```
'b <method name> <bytecode index>'    set breakpoint
```

however `<bytecode index>` is optional and defaults to 0.

madsager, 2015-10-23:
Whoops, that was a random click apparently with the mouse hovering over 'close'. ;-)

madsager, 2015-10-23:
The first argument is actually optional as well and defaults to 'main'.

The documentation for 'bf' is not correct either.

```
'bf <file> <line> <column>'           set breakpoint
```

Line and column default to 1 and there is a version that is not documented:

```
'bf <file> [line] [pattern]'                    set breakpoint at the first occurrence of pattern on the indicated line
```

So for main.dart:

```
0 main() {
1   foo(bar(), baz());
2 }
```

You can do:

```
> bf main.dart 1 baz
```

to set a breakpoint when you get to the evaluation of the call to baz() in the argument list.

closed 2015-10-23 by madsager