dart-archive/sdk #202: Killing the fletch binary may give Bad state error

ricowind, 2015-10-06:
cc @peter-ahe-google @wibling 
I don't know if this is actually intentional, it may very well be, filing this so I don't forget, please just close if WAI. 

OS: Mac
Version 0.1.0-dev.6.0

How to reproduce the issue (this does by far not reproduce on every run):
```
while [ $? -eq 0 ]; do ./bin/fletch run samples/general/hello.dart; done
```
Hit ctrl+c

Expected output (and what happens in most cases):
```
ricow-macbookpro2:fletch-sdk ricow$ while [ $? -eq 0 ]; do ./bin/fletch run samples/general/hello.dart; done
Hello from Darwin running on ricow-macbookpro2.roam.corp.google.com.
Hello from Darwin running on ricow-macbookpro2.roam.corp.google.com.
Hello from Darwin running on ricow-macbookpro2.roam.corp.google.com.
Hello from Darwin running on ricow-macbookpro2.roam.corp.google.com.
Hello from Darwin running on ricow-macbookpro2.roam.corp.google.com.
^CSession 'local' was terminated
```

Actual output:
```
ricow-macbookpro2:fletch-sdk ricow$ while [ $? -eq 0 ]; do ./bin/fletch run samples/general/hello.dart; done
Hello from Darwin running on ricow-macbookpro2.roam.corp.google.com.
Hello from Darwin running on ricow-macbookpro2.roam.corp.google.com.
Hello from Darwin running on ricow-macbookpro2.roam.corp.google.com.
^CBad state: Trying to send command CommandCode.ProcessSpawnForMain() to fletch-vm, but the connection is already closed.
#0      FletchVmSession.sendCommand.<sendCommand_async_body> (package:fletchc/session.dart:124:7)
#1      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#2      _rootRun (dart:async/zone.dart:904)
#3      _CustomZone.run (dart:async/zone.dart:803)
#4      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#5      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#6      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#7      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#8      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

peter-ahe-google, 2015-10-06:
This isn't supposed to happen, and you're also supposed to be asked to file a bug report.

However, this is better than yesterday.

mit-mit, 2015-12-07:
Still repros.

peter-ahe-google, 2015-12-07:
I marked this as started, and I think I may have a CL that I reverted that was intended to fix this. I'll try to investigate more.