dart-archive/sdk #203: Moving the sdk directory will make the compiler break

ricowind, 2015-10-06:
cc @peter-ahe-google @wibling 
OS: mac
Version 0.1.0-dev.6.0

Moving the sdk directory can break the compiler in two ways, depending on if the persistent process is killed or not. This will also reproduce if you try several sdk versions but download them into different directories that you delete after each use.

How to reproduce the first problem:
unzip the sdk bundle to /tmp/fletch-6.0
remove ~/.fletch
remove ~/remote.fletch-settings

do:
```
ricow-macbookpro:tmp ricow$ cd fletch-6.0/fletch-sdk/
ricow-macbookpro:fletch-sdk ricow$ bin/fletch run samples/raspberry_pi/basic/buzzer.dart in session remote
Please enter IP address of remote device: 192.168.2.2
Created settings file 'file:///Users/ricow/remote.fletch-settings'
^Cricow-macbookpro:fletch-sdk ricow$ cd /tmp
ricow-macbookpro:tmp ricow$ mv fletch-6.0 fletch-6
ricow-macbookpro:tmp ricow$ cd fletch-6/fletch-sdk/
ricow-macbookpro:fletch-sdk ricow$ bin/fletch run samples/raspberry_pi/basic/buzzer.dart in session remote
Invalid argument(s): [libraryRoot]: Dart SDK library not found in 'file:///private/tmp/fletch-6.0/fletch-sdk/internal/dart_lib/'.
#0      FletchCompiler.FletchCompiler (package:fletchc/fletch_compiler.dart:128:7)
#1      reuseCompiler.<reuseCompiler_async_body> (package:fletchc/incremental/caching_compiler.dart:60:26)
#2      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#3      _rootRun (dart:async/zone.dart:904)
#4      _CustomZone.run (dart:async/zone.dart:803)
#5      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#6      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#7      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#8      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#9      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

Now, to reproduce the second problem, kill the persistent process, and try again:
```
ricow-macbookpro:fletch-sdk ricow$ pkill dart
ricow-macbookpro:fletch-sdk ricow$ bin/fletch run samples/raspberry_pi/basic/buzzer.dart in session remote
Error: Package config file 'file:///private/tmp/fletch-6.0/fletch-sdk/internal/fletch-sdk.packages' is invalid.
Error reading '/private/tmp/fletch-6.0/fletch-sdk/internal/fletch-sdk.packages' (No such file or directory)
/private/tmp/fletch-6/fletch-sdk/samples/raspberry_pi/basic/buzzer.dart:10:8:
Error: Library not found 'package:gpio/gpio.dart'.
import 'package:gpio/gpio.dart';
       ^^^^^^^^^^^^^^^^^^^^^^^^
/private/tmp/fletch-6/fletch-sdk/samples/raspberry_pi/basic/buzzer.dart:11:8:
Error: Library not found 'package:raspberry_pi/raspberry_pi.dart'.
import 'package:raspberry_pi/raspberry_pi.dart';
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
/private/tmp/fletch-6/fletch-sdk/samples/raspberry_pi/basic/buzzer.dart:19:3:
Warning: Cannot resolve type 'RaspberryPi'.
  RaspberryPi pi = new RaspberryPi();
  ^^^^^^^^^^^
/private/tmp/fletch-6/fletch-sdk/samples/raspberry_pi/basic/buzzer.dart:19:24:
Warning: Cannot resolve 'RaspberryPi'.
  RaspberryPi pi = new RaspberryPi();
                       ^^^^^^^^^^^
/private/tmp/fletch-6/fletch-sdk/samples/raspberry_pi/basic/buzzer.dart:20:3:
Warning: Cannot resolve type 'PiMemoryMappedGPIO'.
  PiMemoryMappedGPIO gpio = pi.memoryMappedGPIO;
  ^^^^^^^^^^^^^^^^^^
/private/tmp/fletch-6/fletch-sdk/samples/raspberry_pi/basic/buzzer.dart:21:24:
Warning: Cannot resolve 'Mode'.
  gpio.setMode(button, Mode.input);
                       ^^^^
/private/tmp/fletch-6/fletch-sdk/samples/raspberry_pi/basic/buzzer.dart:22:25:
Warning: Cannot resolve 'Mode'.
  gpio.setMode(speaker, Mode.output);
                        ^^^^
Compile error
RangeError (index): Invalid value: Not in range 0..4, inclusive: -1
#0      List.[] (dart:core-patch/array.dart:11)
#1      StackTrace.visibleFrame (package:fletchc/stack_trace.dart:153:23)
#2      StackTrace.list (package:fletchc/stack_trace.dart:161:12)
#3      DebugState.list (package:fletchc/debug_state.dart:135:30)
#4      Session.list.<list_async_body> (package:fletchc/session.dart:551:23)
#5      _rootRunUnary (dart:async/zone.dart:911)
#6      _CustomZone.runUnary (dart:async/zone.dart:811)
#7      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#8      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#9      _Future._completeWithValue (dart:async/future_impl.dart:368)
#10     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#11     _rootRun (dart:async/zone.dart:904)
#12     _CustomZone.run (dart:async/zone.dart:803)
#13     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#14     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#15     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#16     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#17     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

peter-ahe-google, 2015-10-07:
I think this should be fixed when we fix #206.

sgjesse, 2015-10-07:
What is causing the problem here is 

```
mv fletch-6.0 fletch-6
```

Which is not #203. In this case the `fletch` command and the compiler (persistent process) have the same version.

peter-ahe-google, 2015-10-08:
The C++ client (fletch) tells the persistent process the absolute path to the fletch program. This path will change if you move the SDK. So I think I can fix this by checking that path. Agreed?

wibling, 2015-11-16:
The 2nd half of the bug (ie. after pkill dart) seems to be fixed. The 1st part is still an issue.