dart-archive/sdk #214: removing power on raspberry pi causes fletch to hang on linux pc

sebe, 2015-10-06:
Rebooting the raspbery pi is fine with the program stopping and returning to the linux prompt,
but when I remove the power from the raspberry pi, I can't break back to the command line with a control-c and any new fletch command hangs without any activity on the pi.
Killing all the fletch processes resolves the issue. 

fletch --version
0.1.0-dev.5.0
uname -a on my linux box.
Linux  3.16.0-38-generic #52~14.04.1-Ubuntu SMP Fri May 8 09:43:57 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

I did get back to the command prompt, once, with this error info, still had to kill the process.

fletch run ./samples/raspberry_pi/basic/knight-rider.dart in session remote
^Z^C^C^C^Z^C^CThe Fletch program is broken and has crashed.

The Fletch team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dart-lang/fletch/issues/new

Please include the following information:

* the name and version of your operating system

* the Fletch SDK version (0.1.0-dev.5.0)

* the entire message you see here (including the full stack trace below)

Uncaught Error: InputError(DiagnosticKind.socketAgentConnectError, {#{message}: No route to host, #{address}: 192.168.178.40:12121})
Stack Trace:
#0      throwFatalError (package:fletchc/src/diagnostic.dart:233:3)
#1      connect.<connect_async_body>.<anonymous closure> (package:fletchc/src/driver/developer.dart:100:9)
#2      _rootRunUnary (dart:async/zone.dart:911)
#3      _CustomZone.runUnary (dart:async/zone.dart:811)
#4      _Future._propagateToListeners.handleError (dart:async/future_impl.dart:525)
#5      _Future._propagateToListeners (dart:async/future_impl.dart:580)
#6      _Future._completeError (dart:async/future_impl.dart:376)
#7      _Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:431)
#8      _rootRun (dart:async/zone.dart:904)
#9      _CustomZone.run (dart:async/zone.dart:803)
#10     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#11     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#12     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#13     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#14     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

#0      _rootHandleUncaughtError.<anonymous closure> (dart:async/zone.dart:895)
#1      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#2      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#3      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#4      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

peter-ahe-google, 2015-10-06:
Thank you for the bug report. This is a duplicate of #188.

peter-ahe-google, 2015-10-06:
BTW: as a work around, run `fletch quit` and try again.

closed 2015-10-06 by peter-ahe-google

peter-ahe-google, 2015-10-07:
I'll reopen this issue. #188 is about the connection to the VM, and this is about the connection to the agent.

peter-ahe-google, 2015-10-07:
@wibling perhaps this is something you should look at? I've developed a mock framework for testing the VM connection, perhaps this can be extended to mock the agent as well. See [CL 1389283002](https://codereview.chromium.org/1389283002/).