dart-archive/sdk #216: fletch-vm should remove any pid and port files created at startup

wibling, 2015-10-06:
If starting the fletch-vm with the --run-dir option it will create a pid and a port file in /var/run/fletch-agent (or more precisely in the directory it reads from the VM_PID_DIR environment variable).

We should clean these up on exit or investigate if we can avoid having these files altogether. E.g. by using sockets/pipes to communicate the information back to the agent when spawning.

mkustermann, 2015-10-06:
If we introduce more files (e.g. `.fletch`, `.fletch.pid`, `fletch-settings`, ...) it would be an option to put these into a common directory (e.g. `$HOME/.fletch/*` or `$HOME/.config/fletch/*`) and have **a single environment variable** like FLETCH_HOME instead of having several environment variables.

@peter-ahe-google

peter-ahe-google, 2015-10-06:
As we want to run the fletch agent locally as well, I think @mkustermann's suggestion makes a lot of sense.

wibling, 2015-11-04:
Instead of adding more files I got rid of the pid and port files.

closed 2015-11-04 by wibling