dart-archive/sdk #229: test.py --help message for runtimes is misleading

ErikCorryGoogle, 2015-10-07:
The test.py --help message for runtimes talks about d8, ff, chrome, safari etc. but these are not actually supported by the test.py script, which has [none, fletchc, fletchvm, fletch_warnings, fletch_tests, fletch_cc_tests] as possibilities.

It does not describe eg the fletch_tests runtime, which is actually supported.

Tested with: python tools/test.py --help

ricowind, 2015-10-07:
Erik: you can easily change the description in tools/testing/dart/test_options.dart

peter-ahe-google, 2015-10-07:
I think we should try to make an effort to use the same version of test.dart as is maintained by the Dart SDK. If so, we should keep changes to a minimum.

sigurdm, 2015-12-10:
A proposed fix is in https://codereview.chromium.org/1505163002/