dart-archive/sdk #230: Moving a function into a patch file causes the compiler to crash

mkustermann, 2015-10-07:
For example doing this change:
```
$ git diff
diff --git a/lib/core/core_patch.dart b/lib/core/core_patch.dart
index 87f8bd3..98a72b5 100644
--- a/lib/core/core_patch.dart
+++ b/lib/core/core_patch.dart
@@ -34,7 +34,7 @@ const patch = "patch";
       throw invocation.asNoSuchMethodError;
     }
     // TODO(ahe): Get rid of this call.
-    fletch.unresolved("<unknown>");
+    _unresolved("<unknown>");
   }
 
   // The noSuchMethod helper is automatically called from the
@@ -664,3 +664,12 @@ const patch = "patch";
     throw new RangeError("The Uri.base path is too large");
   }
 }
+
+_unresolved(name) {
+  throw new NoSuchMethodError(
+      null,
+      name,
+      null,
+      null);
+}
+
diff --git a/lib/system/system.dart b/lib/system/system.dart
index 32183e3..ee15d68 100644
--- a/lib/system/system.dart
+++ b/lib/system/system.dart
@@ -60,14 +60,6 @@ runToEnd(entry) {
   Fiber.exit(entry());
 }
 
-unresolved(name) {
-  throw new NoSuchMethodError(
-      null,
-      name,
-      null,
-      null);
-}
-
 compileError() {
   print("Compile error");
   yield(InterruptKind.compileTimeError.index);
```

Will make fletchc crash with:

```
$ out/ReleaseIA32/fletch run test.dart
/usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/test.dart:
Internal Error: The compiler crashed when compiling this element.

The Fletch compiler is broken.

When compiling the above element, the compiler crashed. It is not
possible to tell if this is caused by a problem in your program or
not. Regardless, the compiler should not crash.

The Fletch team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dart-lang/fletch/issues/new

Please include the following information:

* the name and version of your operating system

* the Fletch SDK version (0.1.0-edge.8675a4f7b911ac06b433ff90545631ed7bf1a7d2)

* the entire message you see here (including the full stack trace
  below as well as the source location above)

The null object does not have a getter 'isAbstractField'.

NoSuchMethodError: method not found: 'isAbstractField'
Receiver: null
Arguments: []
#0      Object._noSuchMethod (dart:core-patch/object_patch.dart:42)
#1      Object.noSuchMethod (dart:core-patch/object_patch.dart:45)
#2      FletchBackend.enqueueHelpers.findHelper (package:fletchc/src/fletch_backend.dart:294:18)
#3      FletchBackend.enqueueHelpers.findExternal (package:fletchc/src/fletch_backend.dart:307:42)
#4      FletchBackend.enqueueHelpers (package:fletchc/src/fletch_backend.dart:321:36)
#5      Compiler.compileLoadedLibraries (package:compiler/src/compiler.dart:1617:13)
#6      Compiler.runCompiler.<anonymous closure> (package:compiler/src/compiler.dart:1529:7)
#7      _rootRunUnary (dart:async/zone.dart:911)
#8      _CustomZone.runUnary (dart:async/zone.dart:811)
#9      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#10     _Future._propagateToListeners (dart:async/future_impl.dart:577)
#11     _Future._completeWithValue (dart:async/future_impl.dart:368)
#12     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#13     _rootRun (dart:async/zone.dart:904)
#14     _CustomZone.run (dart:async/zone.dart:803)
#15     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#16     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#17     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#18     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#19     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

```

peter-ahe-google, 2015-10-07:
The compiler uses the method "unresolved" to generate NSM of static/top-level methods. So it's crashing because it's trying find it and fails to do so.

mkustermann, 2015-10-07:
Yes, but why is it crashing the way it does? Can't it print out an error message that an expected function/class could not be found - instead of a `NoSuchMethodError: method not found: 'isAbstractField'` ?

[I ran into this a while ago and just wanted to file a bug about it.]

peter-ahe-google, 2015-10-07:
We can improve the message, I've changed the subject accordingly.

peter-ahe-google, 2015-10-07:
Proposed fix in [CL 1388393003](https://codereview.chromium.org/1388393003/)

commit b128b92c (2015-10-08): Handle missing helpers better

Fixes #230

R=johnniwinther@google.com

Review URL: https://codereview.chromium.org/1388393003 .
closed 2015-10-08 by peter-ahe-google