dart-archive/sdk #234: The service compiler doesn't allow using the same name for a struct and a service

smanilov, 2015-10-09:
Given an input like

```
service Foo {
}

struct Foo {
}
```

```
stderr:
InternalCompilerError: Multiple definitions for Foo
#0      compileInput.<compileInput_async_body> (package:servicec/compiler.dart:85:7)
#1      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#2      _rootRun (dart:async/zone.dart:900)
#3      _CustomZone.run (dart:async/zone.dart:803)
#4      _CustomZone.runGuarded (dart:async/zone.dart:709)
#5      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#6      _rootRun (dart:async/zone.dart:904)
#7      _CustomZone.run (dart:async/zone.dart:803)
#8      _CustomZone.runGuarded (dart:async/zone.dart:709)
#9      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#10     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#11     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#12     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:394)
#13     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:414)
#14     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:148)
```

smanilov, 2015-11-02:
This is now part of the IDL specification. Need to update wiki...

zerny, 2015-11-06:
Duplicate of resolved issue #256.

closed 2015-11-06 by zerny