dart-archive/sdk #238: Documentation does not indicate which tasks are fast and which are slow

lukechurch, 2015-10-12:
Some of the install instructions take seconds to complete (download the SDK), others tens of minutes (download Raspbian, flash SD card).

This results in users have no sense for how long they need to wait so they can't effectively multitask, or work out when the tasks have failed.

We should provide typical times for each of the tasks that takes over 30s or so, at fairly coarse granularity.

mit-mit, 2015-12-07:
@lukechurch I think we have this now:
http://dart-lang.github.io/fletch/

Downloading: Download of the Raspbian base image: *This will take 5-10 minutes*.

Flashing: Flashing of the image onto the SD card. *This will also take 5-10 minutes*.

Anywhere else this is missing?

sgjesse, 2015-12-08:
Shouldn't this be phrased something like:

This can take up to 5-10 minutes depending on the speed of your Internet connection.

This can take will take 5-10 minutes depending on the speed of your CD card.

Also we have progress indication for both these in the flash_sd_card tool.

commit 17daa25c (2015-12-10): Update getting started instructions

Fixes #238

R=sgjesse@google.com

Review URL: https://codereview.chromium.org/1516483003 .
closed 2015-12-10 by mit-mit