dart-archive/sdk #242: FletchBackend.createDebugInfo triggers assertion error

peter-ahe-google, 2015-10-14:
```
$ ./out/DebugIA32Clang/fletch run tests/corelib/expando_test.dart in session test
third_party/dart/sdk/lib/core/expando.dart:39:3:
Error: External function is not supported
  external Expando([String name]);
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
third_party/dart/sdk/lib/core/expando.dart:53:3:
Error: External function is not supported
  external T operator [](Object object);
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
third_party/dart/sdk/lib/core/expando.dart:62:3:
Error: External function is not supported
  external void operator []=(Object object, T value);
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Compile error
'package:fletchc/src/fletch_backend.dart': Failed assertion: line 739 pos 12: 'Bytecode.identicalBytecodes(expectedBytecodes,
                                       codegen.assembler.bytecodes)' is not true.
#0      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:27)
#1      FletchBackend.createDebugInfo (package:fletchc/src/fletch_backend.dart:739:12)
#2      IncrementalCompiler.createDebugInfo (package:fletchc/incremental/fletchc_incremental.dart:206:38)
#3      DebugState.getDebugInfo.<anonymous closure> (package:fletchc/debug_state.dart:104:31)
#4      _HashVMBase&MapMixin&&_LinkedHashMapMixin.putIfAbsent (dart:collection-patch/compact_hash.dart:254)
#5      DebugState.getDebugInfo (package:fletchc/debug_state.dart:103:23)
#6      StackFrame.debugInfo (package:fletchc/stack_trace.dart:31:41)
#7      StackFrame.list (package:fletchc/stack_trace.dart:34:12)
#8      StackTrace.list (package:fletchc/stack_trace.dart:168:32)
#9      DebugState.list (package:fletchc/debug_state.dart:135:30)
#10     Session.list.<list_async_body> (package:fletchc/session.dart:551:23)
#11     _rootRunUnary (dart:async/zone.dart:911)
#12     _CustomZone.runUnary (dart:async/zone.dart:811)
#13     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#14     _Future._propagateToListeners (dart:async/future_impl.dart:577)
#15     _Future._completeWithValue (dart:async/future_impl.dart:368)
#16     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#17     _rootRun (dart:async/zone.dart:904)
#18     _CustomZone.run (dart:async/zone.dart:803)
#19     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#20     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#21     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#22     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#23     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

peter-ahe-google, 2015-10-14:
Assigned @wibling, but @madsager may also be interested.

madsager, 2015-10-23:
This looks like an issue where compilation has side-effects on the front-end. Peter, do you recall the issue that we had with reporting compile-time errors for disabled bigints and for mirrors not being present. When we report compile-time errors the elements are marked as having errors and therefore the debugger cannot recompile the method to collect debug information. It looks like this is another instance of that. There is a compile-time error leading to a place where we cannot recompile the method for debug info because the first compilation had side-effects on the frontend.

peter-ahe-google, 2015-10-23:
I know how to fix that.