dart-archive/sdk #243: The compilation for the fletch-agent Debian package is using a newer libc version

sgjesse, 2015-10-14:
Right now the fletch agent is compiled with a recent cross-compiler which has lib version 2.19 (as far as I remember). However the Raspberry Pi 2 wheezy image have libc 2.13.

The version checking in the package is currently turned off, as otherwise the `dpkg -i` installation will fail.

One option is to use an alternative sysroot when compiling.

Chromium is compiling using a Debian wheezy sysroot. Look for sysroot in https://chromium.googlesource.com/chromium/src/+/master/build/common.gypi

The sysroots Chromium use are stored in GCS, and the scripts for handling then are here: https://chromium.googlesource.com/chromium/src/+/master/build/linux/sysroot_scripts/

sgjesse, 2015-10-14:
Looks as if the current cross-compiler we are using does not have sysroot support.

Simple `hello.c`

```
main(int argc, char** argv) {
  printf("Hello, world!\n");
}
```

Compiling with `gcc`

```
$ gcc hello.c
```

Now compiling with `gcc` and changing sysroot

```
$ gcc --sysroot=xxx hello.c
In file included from xxx/usr/include/stdio.h:28:0,
                 from hello.c:1:
xxx/usr/include/features.h:323:26: fatal error: bits/predefs.h: No such file or directory
 #include <bits/predefs.h>
                          ^
compilation terminated.
```

However compiling with the cross-compiler `arm-linux-gnueabihf-gcc`, there is no error when passing a non-existent sysroot

```
$ arm-linux-gnueabihf-gcc hello.c
$ arm-linux-gnueabihf-gcc --sysroot=xxx hello.c 
```

However looking at the output from `gcc -v` and  `arm-linux-gnueabihf-gcc -v` both compilers seems to be configured the same way with relation to sysroot.

ricowind, 2015-10-14:
https://bugs.launchpad.net/ubuntu/+source/gcc-defaults-armel-cross/+bug/1375071

ricowind, 2016-02-11:
Since we are now providing a flashable image based on Rasbian Jessie I don't think this is an issue anymore. From my pi with a fresh dartino image:
```
pi@dartino ~ $ ldd --version
ldd (Debian GLIBC 2.19-18+deb8u1) 2.19
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

And on my linux box:
```
nobbo:gcc-cross ricow$ dpkg -s libc6-armhf-cross | grep Version
Version: 2.19-0ubuntu2cross1.104
```

As already discussed we should eventually roll our own version of the cross compiler (to eliminate the need for special setup on bots/dev machines) - but for now I think we are ok with the default cross compiler.

@sgjesse how can I check if my assumption is correct, i.e., that we can install the package without any version checks disabled?

ricowind, 2016-02-11:
I validated that we can reenable the dependency that performs the check, cl here:
https://codereview.chromium.org/1682193005

commit 792f16af (2016-02-11): Enable check for correct libc version in raspbian agent package

After moving to Jessie based images we no longer have issues with libc versions.

Closes #243

R=sgjesse@google.com
BUG=

Review URL: https://codereview.chromium.org/1682193005 .
closed 2016-02-11 by ricowind