dart-archive/sdk #244: The fletch driver crashes when supplied with a '-D' option

smanilov, 2015-10-14:
When the fletch driver is given any option starting with '-D' it crashes. Example:

```
fletch -Dcrash
```

outputs

```
The Fletch program is broken and has crashed.

The Fletch team would greatly appreciate if you would take a moment to
report this problem at https://github.com/dart-lang/fletch/issues/new

Please include the following information:

* the name and version of your operating system

* the Fletch SDK version (0.1.0-edge.5418ca05d3511d83e8dcf6a2416c37dda29211aa)

* the entire message you see here (including the full stack trace below)

The null object does not have a getter 'name'.

NoSuchMethodError: method not found: 'name'
Receiver: null
Arguments: []
#0      Object._noSuchMethod (dart:core-patch/object_patch.dart:42)
#1      Object.noSuchMethod (dart:core-patch/object_patch.dart:45)
#2      Options.parse (package:fletchc/src/verbs/options.dart:216:47)
#3      ClientController.parseArguments (package:fletchc/src/driver/driver_main.dart:445:31)
#4      handleVerb.<handleVerb_async_body>.performVerb.<performVerb_async_body> (package:fletchc/src/driver/driver_main.dart:262:12)
#5      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#6      _rootRun (dart:async/zone.dart:900)
#7      _CustomZone.run (dart:async/zone.dart:803)
#8      _CustomZone.runGuarded (dart:async/zone.dart:709)
#9      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#10     _rootRun (dart:async/zone.dart:904)
#11     _CustomZone.run (dart:async/zone.dart:803)
#12     _CustomZone.runGuarded (dart:async/zone.dart:709)
#13     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#14     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#15     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#16     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:394)
#17     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:414)
#18     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:148)
```

I would expect it to handle the option and exit gracefully if it is not supported.

smanilov, 2015-10-15:
The problem is in pkg/fletchc/lib/src/verbs/options.dart

```
141|      var parsedArgument;
142|      if (optionString.startsWith("-D")) {
143|        // Define.
144|        kind = OptionKind.define;
145|      } else if (optionString.startsWith("-")) {
...|      ...
192|      }
193|
194|      switch (kind) {
...|      ...
214|        case OptionKind.define:
215|          testDebuggerCommands[parsedArgument.name] = parsedArgument.value;
216|          break;
```

When '-D' is passed parsedArgument will be always `null` when it is used on line 215.

wibling, 2015-11-09:
This is not critical for 0.2.0 so moving it out.

mit-mit, 2015-12-07:
@wibling are you working on this, or should we unassign it so that others can poach it?

wibling, 2015-12-07:
I am not working on it. I have unassigned it.

commit 1d01ba4c (2015-12-10): Properly store defines passed via the command line.

Fixes #244.

BUG=
R=wibling@google.com

Review URL: https://codereview.chromium.org/1509013006 .
closed 2015-12-10 by herhut-ggl