dart-archive/sdk #246: If Coroutine._coroutineStart is missing, the VM crashes

peter-ahe-google, 2015-10-15:
The native `CoroutineNewStack` attempts to invoke `_coroutineStart`, but doesn't go through NSM if the method is missing. Instead it crashes.

I think there's a similar problem in `ProcessSpawn`:

```
  Function* entry = FunctionForClosure(entrypoint, 2);
  ASSERT(entry != NULL);
```

Would it be problematic to invoke NSM or otherwise report an error?

peter-ahe-google, 2015-10-15:
@ErikCorryGoogle @zerny may be interested in this.

skabet, 2015-10-28:
We depend on the compilers output, on shield off this specific code. Unless this model changes, this issues is, IMO, working-as-intended.

(should me noted, as we discussed offline, a crash here can only happen if the compiler or our library patch codes, is modified)

peter-ahe-google, 2015-10-29:
That's right, @skabet. My biggest concern is that this is brittle and likely to cause issues as we move to a larger team and it isn't the same individuals that work on libraries and natives. If there's a good reason for using a brittle lookup mechanism, then we shouldn't do anything about this. However, I suspect that strengthening this code is easy.