dart-archive/sdk #248: fletch_tests/service_tests/performance is flaky

peter-ahe-google, 2015-10-15:
From [build bot](http://build.chromium.org/p/client.fletch/builders/fletch-linux-asan-x86/builds/11/steps/test_ReleaseIA32Asan%20failures/logs/stdio):

```
FAILED: none-fletch_tests release_ia32_asan fletch_tests/service_tests/performance
Expected: Pass 
Actual: Timeout
CommandOutput[fletch_test]:

stdout:
Running:  tools/cxx_wrapper.py -DFLETCH_ASAN -L/FLETCH_ASAN -std=c++11 -DFLETCH_ENABLE_FFI -DFLETCH_ENABLE_LIVE_CODING -DFLETCH_ENABLE_PRINT_INTERCEPTORS -DFLETCH_ENABLE_NATIVE_PROCESSES -m32 -DFLETCH32 -DFLETCH_TARGET_IA32 -DFLETCH_TARGET_OS_LINUX -DFLETCH_TARGET_OS_POSIX -I. -Lthird_party/libs/linux/x86 -o out/ReleaseIA32Asan/generated_service_tests/performance/service_performance_test -Wl,--start-group tests/service_tests/performance/performance_test.cc tests/service_tests/performance/cc/performance_service.cc tests/service_tests/performance/cc/struct.cc tests/service_tests/performance/cc/unicode.cc out/ReleaseIA32Asan/libfletch.a -Wl,--end-group -ltcmalloc_minimal -lpthread -ldl -rdynamic
stdout:

stderr:

Running:  out/ReleaseIA32Asan/generated_service_tests/performance/service_performance_test out/ReleaseIA32Asan/generated_service_tests/performance/performance.snapshot

Command[fletch_test]: FletchTestCommand(service_tests/performance)
Took 0:01:00.042107

Short reproduction command (experimental):
    python tools/test.py -mrelease --kill-persistent-process=0 --run-gclient-hooks=0 --build-before-testing=0 --asan -rfletch_tests -t60 fletch_tests/service_tests/performance
```

peter-ahe-google, 2015-10-15:
@smanilov @zerny

zerny, 2015-11-06:
I have not been able to reproduce any timeouts with this test.

closed 2015-12-07 by mit-mit