dart-archive/sdk #250: Timeout  issues when running `./tools/test.py`

mkustermann, 2015-10-19:
Sometimes when running tests, I get error messages about timeouts and tests will be reported as `Crash`:

```
$ ./tools/test.py unsorted
...
[00:04 |   2% | +    2 | -    0]Timed out: out/DebugIA32/fletch create session 2 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 4 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 6 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 8 with .fletch-settings

FAILED: none-fletchc debug_ia32 unsorted/string_from_iterable_test
Expected: Pass 
Actual: Crash
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugIA32/fletch create session 4 with .fletch-settings

 => 253
$ out/DebugIA32/fletch x-end session 4

 => 254

stderr:
$ out/DebugIA32/fletch create session 4 with .fletch-settings
Session '4' was terminated
Fletch VM exitcode is null
Non-zero exit code (253) from: out/DebugIA32/fletch create session 4 with .fletch-settings
$ out/DebugIA32/fletch x-end session 4
Couldn't find a session called '4'. Try running 'fletch create session 4'
Non-zero exit code (254) from: out/DebugIA32/fletch x-end session 4

Command[fletch_session]: 

There are three ways to reproduce this error:

  1. Run the test exactly as in this test framework. This is the hardest to
     debug using gdb:

    /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tools/testing/bin/linux/dart -c -Dtest.fletch_session_comma
nd.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
       tools/testing/dart/fletch_session_command.dart out/DebugIA32/fletch \
       /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/unsorted/string_from_iterable_test.dart

  2. Run the helper program `tests/fletchc/run.dart` under `gdb` using
     `follow-fork-mode child`. This can be confusing, but makes it easy to run
     a reproduction command in a loop:

    gdb -ex 'follow-fork-mode child' -ex run --args \
        out/DebugIA32/dart -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settin
gs -c tests/fletchc/run.dart \
        /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/unsorted/string_from_iterable_test.dart

  3. Run the `fletch-vm` in gdb and attach to it via the helper program. This
     is the easiest way to debug using both gdb and lldb. You need to start two
     processes, each in their own terminal window:

    gdb -ex run --args out/DebugIA32/fletch-vm --port=54321

    out/DebugIA32/dart -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings -
c -DattachToVm=54321 \
      tests/fletchc/run.dart /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/unsorted/string_from_iterabl
e_test.dart

Took 0:00:22.161085

Short reproduction command (experimental):
    python tools/test.py -rfletchc -t120 unsorted/string_from_iterable_test

```

I assume this was somehow caused by [Improve handling of time out and compiler crashes](https://codereview.chromium.org/1403003005).

Increasing the timeout from `20` to `60` seconds, fixes usually the issue, i.e.:
```diff
diff --git a/tools/testing/dart/fletch_session_command.dart b/tools/testing/dart/fletch_session_command.dart
index 5ee31ea..a8bd2fa 100644
--- a/tools/testing/dart/fletch_session_command.dart
+++ b/tools/testing/dart/fletch_session_command.dart
@@ -56,7 +56,7 @@ const String settingsFileName =
 /// Default timeout value (in seconds) used for running commands that are
 /// assumed to complete fast.
 // TODO(ahe): Lower this to 5 seconds.
-const int defaultTimeout = 20;
+const int defaultTimeout = 60;
```

@peter-ahe-google Maybe you can take a look?

peter-ahe-google, 2015-10-19:
Thanks for reporting this. The details are pretty useful, and I have a pretty good idea how to fix it.

I'm reaching out to you on chat for some additional information.

peter-ahe-google, 2015-10-19:
Chatting with Martin confirmed my suspicion: we're starting too many isolates at the same time. So I believe the work around is to use "-j16" when running test.py (if you have 16 cores with hyper-threading the default is -j32).

peter-ahe-google, 2015-10-22:
My bad: I haven't yet landed [CL 1403763007](https://codereview.chromium.org/1403763007) which was supposed to have landed before [CL1403003005](https://codereview.chromium.org/1403003005).

So the issue isn't that the VM can't start 32 isolates at the same time in 20 seconds. The issue is that it takes more than approximately 667ms (20/30) to start one isolate.

mkustermann, 2015-11-04:
@peter-ahe-google  Even though all of your CLs seem to have landed now, I still get the issue somteimes at revision 05a4392cf78ea890c8a00bd4739b726f13b2c69b with `./tools/test.py -cfletchc -rfletchvm`.

peter-ahe-google, 2015-11-05:
Could you include a log if it happens again?

mkustermann, 2015-11-05:
Sure, but there is no change, the behaviour is the same. This is running at a51e673b28e0a93256f43d71d427cc347fd0473f:

```
$ ./tools/test.py -cfletchc -rfletchvm
....
________ running 'ninja -C fletch' in '/usr/local/google/home/kustermann/repositories/fletch-gclient2'
ninja: Entering directory `fletch'
[1/2] python tools/update_version_gyp.py version.gyp
[2/2] GYP_CROSSCOMPILE=1 ./third_party/gyp/gyp --depth=. -Idefault_targets.gypi -Goutput_dir=out --format=ninja fletch.gyp
Test configuration: fletchc_fletchvm_debug_ia32
Building in out/DebugIA32.
Use --build-before-testing=0 to skip this step.
Building done.
Total: 4474 tests
 * 20 tests will be skipped (1 skipped by design)
 * 0 tests are expected to be flaky but not crash
 * 3492 tests are expected to pass
 * 1 tests are expected to fail that we won't fix
 * 926 tests are expected to fail that we should fix
 * 1 tests are expected to crash that we should fix
 * 0 tests are allowed to timeout
 * 0 tests are skipped on browsers due to compile-time error

Timed out: out/DebugIA32/fletch create session 1 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 2 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 3 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 4 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 6 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 8 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 11 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 13 with .fletch-settings
Timed out: out/DebugIA32/fletch create session 14 with .fletch-settings
[00:26 |   0% | +    2 | -    0]
FAILED: fletchc-fletchvm debug_ia32 coroutine/yield_test
Expected: Pass 
Actual: Crash
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugIA32/fletch create session 1 with .fletch-settings

 => 253
$ out/DebugIA32/fletch x-end session 1

 => 254

stderr:
$ out/DebugIA32/fletch create session 1 with .fletch-settings
Session '1' was terminated
Fletch VM exitcode is null
Non-zero exit code (253) from: out/DebugIA32/fletch create session 1 with .fletch-settings
$ out/DebugIA32/fletch x-end session 1
Couldn't find a session called '1'. Try running 'fletch create session 1'
Non-zero exit code (254) from: out/DebugIA32/fletch x-end session 1

Command[fletch_session]: 

There are three ways to reproduce this error:

  1. Run the test exactly as in this test framework. This is the hardest to
     debug using gdb:

    /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tools/testing/bin/linux/dart -c -Dtest.fletch_session_comma
nd.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
       -Dfletch.version=`out/DebugIA32/fletch-vm --version` \
       tools/testing/dart/fletch_session_command.dart out/DebugIA32/fletch \
       /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/coroutine/yield_test.dart

  2. Run the helper program `tests/fletchc/run.dart` under `gdb` using
     `set follow-fork-mode child`. This can be confusing, but makes it
     easy to run a reproduction command in a loop:

    gdb -ex 'set follow-fork-mode child' -ex run --args \
        out/DebugIA32/dart -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settin
gs \
        -Dfletch.version=`out/DebugIA32/fletch-vm --version` \
        -c tests/fletchc/run.dart /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/coroutine/yield_test.da
rt

  3. Run the `fletch-vm` in gdb and attach to it via the helper program. This
     is the easiest way to debug using both gdb and lldb. You need to start two
     processes, each in their own terminal window:

    gdb -ex run --args out/DebugIA32/fletch-vm --port=54321

    out/DebugIA32/dart -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
      -Dfletch.version=`out/DebugIA32/fletch-vm --version` \
      -c -DattachToVm=54321 tests/fletchc/run.dart /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/corout
ine/yield_test.dart

Took 0:00:20.713271

Command[vm]: out/DebugIA32/fletch-vm /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/out/DebugIA32/generated_co
mpilations/fletchc/tests_coroutine_yield_test/fletch.snapshot
Did not run

Command[vm]: out/DebugIA32/fletch-vm -Xunfold-program /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/out/Debug
IA32/generated_compilations/fletchc/tests_coroutine_yield_test/fletch.snapshot
Did not run

Short reproduction command (experimental):
    python tools/test.py -cfletchc -rfletchvm -t120 coroutine/yield_test

...
```

peter-ahe-google, 2015-11-05:
Thank you, I agree this is the same problem. I'm not able to reproduce this locally, perhaps you can help me debug on your machine tomorrow?

mkustermann, 2015-11-06:
After debugging this we can conclude that the DartVM does blocking spawning by loading all fletchc sources again for each new isolate. This basically serializes all isolate spawning on the main thread and takes roughly 16 seconds on a 32 core machine.

This is a serious issue with the DartVM and we've filed https://github.com/dart-lang/sdk/issues/24849 and hope it gets fixed.

The only workarounds I can think of are:
  * Make a snapshot of the persistent process and use that. It seems then isolate spawning goes at least down to **100ms**
  * Use a very small jump-isolate via `Isolate.spawnUri` which can quickly load and spawn from there the real compiler isolate.

peter-ahe-google, 2015-11-06:
I still have reservations around using Dart VM snapshots because of this issue: [https://github.com/dart-lang/sdk/issues/11458].

However, we should look more closely at Dart VM snapshots and see what the pros and cons are for performance.

Using `Isolate.spawnUri` would require us to rewrite how verbs run tasks in a worker isolate, but that should be easy.

peter-ahe-google, 2015-11-06:
Also, thank you so much for helping out getting a deeper understanding of this issue!

mit-mit, 2015-12-07:
@mkustermann have we seen this after we increased the number of file descriptors on the Mac bots?

mit-mit, 2015-12-07:
@ricowind FYI

mkustermann, 2015-12-08:
AFAIK Our version of the DartVM has been rolled to a version that fixes the blocking isolate spawning issue (i.e. it happens in the background).

This made us run out of file descriptors on MacOS (due to parallel isolate spawning and re-reading sources by the DartVM), which we also fixed.

So the original issue I reported should be fixed and personally I've not seen this particular timeout issue.

@peter-ahe-google  If you agree, then we can close this issue.

mit-mit, 2015-12-09:
Closing, @peter-ahe-google re-open if you do not agree.

closed 2015-12-09 by mit-mit