dart-archive/sdk #252: ForeignLibrary.fromName affects ForeignLibrary.main

smanilov, 2015-10-21:
Consider the adding the following function in `tests/ffi/ffi_test.dart`:

```
void testBug() {
  var libPath = ForeignLibrary.bundleLibraryName('ffi_test_library');
  ForeignLibrary fl = new ForeignLibrary.fromName(libPath);
  var memuint32 = fl.lookup('memuint32');
}
```

This doesn't throw an error, as expected: `memuint32` can be found in `ffi_test_library`.

Now change the object of the lookup to `ForeignLibrary.main`:

```
void testBug() {
  var libPath = ForeignLibrary.bundleLibraryName('ffi_test_library');
  ForeignLibrary fl = new ForeignLibrary.fromName(libPath);
  var memuint32 = ForeignLibrary.main.lookup('memuint32');
}
```

Surprisingly, this still works, even though it shouldn't: `memuint32` is not part of main. Verify by commenting out the first two lines of the function:

```
void testBug() {
//  var libPath = ForeignLibrary.bundleLibraryName('ffi_test_library');
//  ForeignLibrary fl = new ForeignLibrary.fromName(libPath);
  var memuint32 = ForeignLibrary.main.lookup('memuint32');
}
```

This fails with an argument error:

```
Uncaught exception: Instance of 'ArgumentError' {
  ArgumentError._hasValue: false
  ArgumentError.invalidValue: null
  ArgumentError.name: null
  ArgumentError.message: null
}
```

ricowind, 2015-10-21:
This is not surprising, since we do dlopen with RTLD_GLOBAL.

From dlopen man page:
 If filename is a NULL pointer, then the returned handle is for the main program.  When given to dlsym(), this handle causes a search for a symbol in the main program, followed by all shared
       libraries loaded at program startup, and then all shared libraries loaded by dlopen() with the flag RTLD_GLOBAL.

The reason for using RTLD_GLOBAL is to allow loading several libraries that depends on each other - we could do this differently by requiring that these are loaded automatically by the linker (this requires setting paths, like requiring the user to use LD_LIBRARY_PATH or another alternative)

I am closing this as working as intended, but I am happy to discuss if this is the right solution in general.

mkustermann, 2015-10-21:
I think that this is confusing from a developer point of view, because `ForeignLibrary.main.lookup('xxx')` will sometimes work and sometimes not. It entirely depends on side-effects from `ForeignLibrary.fromName()`.

One option would be to let the user decide if a library he is loading should be used for future symbol resolution from main.

ricowind, 2015-10-21:
Yes, that is actually a good suggestion, let me do that

mkustermann, 2015-10-21:
I begin a bit to question RTLD_GLOBAL in general, since the manpage says:

External references in the library are resolved using the libraries **in that library's dependency list** and any other libraries previously opened with the RTLD_GLOBAL flag.

So the depended libraries should be loaded if compiled with the right flags. See also an interessting discussion about lua: http://lua-users.org/lists/lua-l/2015-05/msg00297.html

ricowind, 2015-10-22:
>> So the depended libraries should be loaded if compiled with the right flags.
That is true, but imagine this: you compile some shared library foo that depends on shared library bar. You put both of these in a bundle next to each other. This will _not_ be in any default linker lookup location. We could have it like this:
my_deploy_folder
  fletch-vm
  app.snapshot
  lib
    libfoo.so
    libbar.so

Loading the foo lib will fail, since it can't find libraries in the dependency list (our lib dir is not in the default search path for libs). Note that I may not have the ability to compile these .so files myself. 

The user could go about solving this by running the vm with LD_LIBRARY_PATH=lib, which would in some cases be more sane (since now I only have to load my foo library and everything works). I am happy to discuss this in detail offline if you have other suggestions.

commit a2dd51a7 (2015-12-07): Be explicit about when to load a library as RTLD_GLOBAL, default to RTLD_LOCAL.

We switched over to use RTLD_GLOBAL to allow subsequent library loads to use already loaded shared libraries. This can cause confusion, see issue 252

Closes #252

R=kustermann@google.com, stanm@google.com

Review URL: https://codereview.chromium.org/1406103003 .
commit 1224b648 (2015-12-09): Be explicit about when to load a library as RTLD_GLOBAL, default to RTLD_LOCAL.

We switched over to use RTLD_GLOBAL to allow subsequent library loads to use already loaded shared libraries. This can cause confusion, see issue 252

Closes #252

R=kustermann@google.com, stanm@google.com

Committed: https://github.com/dart-lang/fletch/commit/a2dd51a7f3161736ce51d440109c90fa12a8961a

Review URL: https://codereview.chromium.org/1406103003 .
closed 2015-12-07 by ricowind