dart-archive/sdk #255: Missing option to add pull up or down to gpio pin

butlermatt, 2015-10-25:
When using a gpio pin for input, particularly with switches, the value may sometimes be floating and may read true when there is no input even connected. 

A solution is often to add a pull up or pull down resistor to the circuit and because it's so common, they are integrated into the Pi directly. The python sdk has an optional parameter to indicate if a pull up or down should be applied to the pin when it is set up. This is currently missing in the dart sdk and could greatly simplify circuits.

sgjesse, 2015-10-26:
If you use the `PiMemoryMappedGPIO` in the `raspberry_pi`package there is a way to use the internal pull up or pull down resistor.

See https://github.com/dart-lang/fletch/blob/master/pkg/raspberry_pi/lib/raspberry_pi.dart#L242.

(looking at the comment I can see it it was not updated after I changed an into to a enum).

It might be that the sysfs driver for the Raspberry Pi also support setting the pull up or pull down resistor.

sgjesse, 2015-11-02:
Having looked further into this I don't think there is a way to set up the built-in pull up or pull down resistors through the sysfs interface.

If the built-in pull up or pull down resistors are used I think the best solution is to make sure they are configured before actually attaching anything to the pins.

butlermatt, 2015-11-02:
For using sysfs I found:
To set pull-up
```echo 0 > /sys/class/gpio/gpio<pin#>/active_low```

mit-mit, 2015-11-13:
tentatively including in 0.3

sgjesse, 2015-12-09:
I am pretty sure that `/sys/class/gpio/gpio<pin#>/active_low` does not affect the internal pull-up/pull-down. The only way to access the pull-up/pull-down are through the memory mapped API.

closed 2015-12-09 by sgjesse