dart-archive/sdk #256: Same-named service and struct cause servicec to generate invalid code

smanilov, 2015-10-26:
Consider the following:

```
service Foo {
}

struct Foo {
    int32 foo;
}
```

This will result in `cc` output containing:

```
class Foo {
   ...
}

class Foo : public Reader {
   ...
}
```

This should not be the case - unique class names should be generated instead.

smanilov, 2015-10-26:
Regression because of https://github.com/dart-lang/fletch/issues/234

zerny, 2015-10-27:
How about we let the definition of "service Foo { ... }" be "class FooService { ... }". The struct definitions will keep the same name. We should then add a check such that "struct FooService { }" is considered invalid.

smanilov, 2015-11-02:
Resolved in https://github.com/dart-lang/fletch/commit/5bf1afe1b70ecf07efe349caede4390c727e5bbc

closed 2015-11-02 by smanilov