dart-archive/sdk #259: Default packages and sessions in the fletch command-line tool

zerny, 2015-10-27:
Here are a few comments after using the fletch tool for running a simple program. I do not mean to imply that the behavior I expected should be the actual behavior. This issue is meant to provide input as to how we might improve the interface.

I expected:
```
$ cd src/myprog
$ ls -a
myprog.dart .packages
$ fletch run myprog.dart
```
to use the `.packages` file relative to `myprog.dart`, ie, `src/myprog/.packages`.

I expected
```
$ fletch create session myprog with myprog.settings
$ fletch run myprog.dart
```
to run in session `myprog`.

I expected
```
$ fletch run myprog.dart in session myprog
$ fletch run
```
to perform the second run with the implicit state of `myprog.dart` and session `myprog`.

mit-mit, 2015-12-07:
@wibling did you fix this?

wibling, 2015-12-07:
Nope. All of the above are still present. This is related to the new Fletch
project idea.

Basically it doesn't make sense to fix these if we agree we should have a
Fletch Project.

mit-mit, 2015-12-09:
Closing this, project proposal is #359

zerny, 2015-12-15:
I don't see how this is addressed in the project proposal. Even when working in a project we need to determine what the current session is and, as an interface for humans, I'd expect the latest session to be reused. If not we should have an explicit reason and consistent UX based on that reason.