dart-archive/sdk #260: Missing fletch-sdk.packages when running from checkout

zerny, 2015-10-27:
Running fletch from a source checkout results in a compiler error being issued for the missing fletch-sdk.packages:

```
$ fletch run main.dart
Error: Package config file 'file:///___/fletch/out/DebugIA32/fletch-sdk.packages' is invalid.
Error reading '/___/fletch/out/DebugIA32/fletch-sdk.packages' (No such file or directory)
/path/to/prog/main.dart:1:8:
```

mit-mit, 2015-12-07:
@wibling this is fixed, right?

wibling, 2015-12-07:
It might be. I am not sure what the exact scenario is?

@zerny, could you elaborate a bit about the repro? Is it having a directory (e.g. myprog) outside of the repo with source and running fletch inside the source directory?

closed 2016-05-30 by mit-mit