dart-archive/sdk #267: fibo(29) is 5 times slower than deltablue(100)

cpu-chromium, 2015-10-29:
This is from the friendly land of SoD.  I don't really know if this is a bug, but looking at the code seems strange, unless there is no tail call elimination, in which case then I am surprised it can recurse so deep without blowing up the stack.

fibonacci.dart:
main() {
  // Expects to return 0.
  return fibonacci(29) - 514229;
}
int fibonacci(int n) {
  return n < 2 ? n : (fibonacci(n - 1) + fibonacci(n - 2));
}

We use the high res (usec) wall timer, like this:

  lk_bigtime_t start = current_time_hires();
  int result = FletchRunMain(program);
  lk_bigtime_t elapsed = current_time_hires() - start;
  printf("fletch-vm ran for %llu usecs\n", elapsed);

Results:

fletch deltablue.snap
ready for deltablue.snap over tftp (at 0x80420000)
] starting fletch-vm...
loading snapshot: 39560 bytes ...
running program...
fletch-vm ran for 182084 usecs
tearing down fletch-vm...
vm exit code: 0

fletch fibonacci.snap
ready for fibonacci.snap over tftp (at 0x804e0000)
] starting fletch-vm...
loading snapshot: 31689 bytes ...
running program...
fletch-vm ran for 1122143 usecs
tearing down fletch-vm...
vm exit code: 0

cpu-chromium, 2015-10-29:
https://github.com/domokit/sod/tree/master/dart/examples

madsager, 2015-10-29:
No tail call elimination (which makes debugging work for tail-recursive
programs). Neither the Dart VM nor V8 have tail call elimination either.
That doesn't matter though because fib is not tail-recursive. You always
need the result of the recursive calls for further computation.
Additionally, the stack doesn't grow that deep here.

I'm surprised about these results as well. We should investigate it
further. On my laptop, which is obviously a different beast) the difference
is big in the other direction:

$ out/ReleaseIA32/fletch export ./benchmarks/DeltaBlue.dart to
deltablue.snapshot
$ time ./out/ReleaseIA32/fletch-vm deltablue.snapshot
DeltaBlue(RunTime): 39341 us.

real    0m2.128s
user    0m2.124s
sys     0m0.004s

$ out/ReleaseIA32/fletch export fib.dart to fib.snapshot
$ time ./out/ReleaseIA32/fletch-vm fib.snapshot

real    0m0.032s
user    0m0.032s
sys     0m0.000s

Cheers,
Mads

madsager, 2015-10-29:
Oh, there is no mystery here. I ran the DeltaBlue with warmup and iterating
for 2 seconds. Downloading the version of deltablue from the sod repo it
looks a lot different. That version does next to nothing:

$ ./out/ReleaseIA32/fletch export deltablue.dart to deltablue-sod.snapshot
$ time ./out/ReleaseIA32/fletch-vm deltablue-sod.snapshot

real 0m0.006s
user 0m0.004s
sys 0m0.000s

$ out/ReleaseIA32/fletch export fib.dart to fib.snapshot
$ time ./out/ReleaseIA32/fletch-vm fib.snapshot

real 0m0.032s
user 0m0.032s
sys 0m0.004s

I modified Fletch to dump the bytecodes that are executed at runtime. Which
also shows the difference. I dump trace files printing a byte code name per
line for each executed bytecode:

$ tail -n 2 fib.trace
LoadLocal1
ProcessYield

Fib 29 just does a lot more than that version of delta blue:

$ wc -l fib.trace
29953456 fib.trace

$ wc -l delta-blue.trace
2364144 delta-blue.trace

And looking at just invokes:

$ grep Invoke fib.trace | wc -l
14144680

$ grep Invoke delta-blue.trace | wc -l
721584

So, yeah, delta blue is a lot of code but it doesn't do a whole lot at
runtime compared to fib(29).

Cheers,
Mads

cpu-chromium, 2015-10-29:
Thanks for the investigation.  The difference between the two deltablue should be minimal, I return in some failure cases. I don't use the test harness so I guess that might be the difference.

I did code a tail recursive fibo in dart and it is very fast, like 600 microseconds, based on http://code.runnable.com/U44bs7vG_3wKYXUm/tail-recursive-fibonacci-c%2B%2B-for-recursion

closed 2015-10-29 by cpu-chromium