dart-archive/sdk #271: Improve test failure reporting from ./tools/test.py test failures

mkustermann, 2015-10-30:
We've seen the following failure. It is impossible for me to find out what went wrong. It says `Fletch VM exitcode is 0`. Why did the test fail then?

Please note that the compile-time error and uncaught exceptions are normal. These are not the problem.

@peter-ahe-google Maybe you have some ideas how we could get better error reporting?

```
FAILED: fletchc-fletchvm debug_ia32_clang_asan unsorted/process_linking_test
Expected: Pass 
Actual: RuntimeError
Unexpected compile-time error.
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugIA32ClangAsan/fletch create session 4 with .fletch-settings

 => 0
$ out/DebugIA32ClangAsan/fletch show log in session 4
[4: 2015-10-29 10:42:11.926 created session with file:///b/build/slave/fletch-linux-debug-asan-x86/build/fletch/.fletch-settings Settings(packages: file:///b/build/slave/fletch-linux-debug-asan-x86/build/fletch/.packages, options: [], constants: {}, device_address: null, device_type: DeviceType.mobile)]

 => 0
$ out/DebugIA32ClangAsan/fletch-vm &
$ out/DebugIA32ClangAsan/fletch attach tcp_socket 127.0.0.1:38137 in session 4

 => 0
$ out/DebugIA32ClangAsan/fletch export /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/tests/unsorted/process_linking_test.dart to file /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/out/DebugIA32ClangAsan/generated_compilations/fletchc/tests_unsorted_process_linking_test/fletch.snapshot in session 4
/b/build/slave/fletch-linux-debug-asan-x86/build/fletch/tests/unsorted/process_linking_test.dart:189:3:
Error: Expected ';' after this.
  a b c;
  ^
/b/build/slave/fletch-linux-debug-asan-x86/build/fletch/tests/unsorted/process_linking_test.dart:189:3:
Warning: Cannot resolve 'a'.
  a b c;
  ^
/b/build/slave/fletch-linux-debug-asan-x86/build/fletch/tests/unsorted/process_linking_test.dart:189:5:
Warning: Cannot resolve type 'b'.
  a b c;
    ^

 => 0
$ out/DebugIA32ClangAsan/fletch x-end session 4
Ended session '4'.

 => 0
$ out/DebugIA32ClangAsan/fletch-vm

stderr:
Fletch VM exitcode is 0

CommandOutput[vm]:

stdout:
Compile error
Uncaught exception:
a OneByteString: "failing"

Compile error
Uncaught exception:
a OneByteString: "failing"

Compile error
Uncaught exception:
a OneByteString: "failing"

Compile error
Uncaught exception:
a OneByteString: "failing"

Compile error
Uncaught exception:
a OneByteString: "failing"

Command[fletch_session]: 

There are three ways to reproduce this error:

  1. Run the test exactly as in this test framework. This is the hardest to
     debug using gdb:

    /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/tools/testing/bin/linux/dart -c -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
       -Dfletch.version=`out/DebugIA32ClangAsan/fletch-vm --version` \
       tools/testing/dart/fletch_session_command.dart out/DebugIA32ClangAsan/fletch \
       /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/tests/unsorted/process_linking_test.dart

  2. Run the helper program `tests/fletchc/run.dart` under `gdb` using
     `set follow-fork-mode child`. This can be confusing, but makes it
     easy to run a reproduction command in a loop:

    gdb -ex 'set follow-fork-mode child' -ex run --args \
        out/DebugIA32ClangAsan/dart -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
        -Dfletch.version=`out/DebugIA32ClangAsan/fletch-vm --version` \
        -c tests/fletchc/run.dart /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/tests/unsorted/process_linking_test.dart

  3. Run the `fletch-vm` in gdb and attach to it via the helper program. This
     is the easiest way to debug using both gdb and lldb. You need to start two
     processes, each in their own terminal window:

    gdb -ex run --args out/DebugIA32ClangAsan/fletch-vm --port=54321

    out/DebugIA32ClangAsan/dart -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
      -Dfletch.version=`out/DebugIA32ClangAsan/fletch-vm --version` \
      -c -DattachToVm=54321 tests/fletchc/run.dart /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/tests/unsorted/process_linking_test.dart

Took 0:00:01.153998

Command[vm]: out/DebugIA32ClangAsan/fletch-vm /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/out/DebugIA32ClangAsan/generated_compilations/fletchc/tests_unsorted_process_linking_test/fletch.snapshot
Took 0:00:00.098000

Command[vm]: out/DebugIA32ClangAsan/fletch-vm -Xunfold-program /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/out/DebugIA32ClangAsan/generated_compilations/fletchc/tests_unsorted_process_linking_test/fletch.snapshot
Did not run

Short reproduction command (experimental):
    python tools/test.py --kill-persistent-process=0 --run-gclient-hooks=0 --build-before-testing=0 -cfletchc -rfletchvm --asan --clang -t120 unsorted/process_linking_test
```

peter-ahe-google, 2015-10-30:
This is tricky. What would have helped here might have been eliding the lines between: 

    CommandOutput[fletch_session]

and

    CommandOutput[vm]:

Because that command is actually working fine (it generates the snapshot as it should). The problem here is that when we run `fletch-vm` on the snapshot, the exit code isn't zero.

However, I don't think it would be a good idea to hide the log from the first command. My suggestion would be to change the format of the error message to something like this:

```
FAILED: fletchc-fletchvm debug_ia32_clang_asan unsorted/process_linking_test
Expected: Pass 
Actual: RuntimeError

The command `out/DebugIA32ClangAsan/fletch-vm /b/build/slave/fletch-linux-debug-asan-x86/build/fletch/out/DebugIA32ClangAsan/generated_compilations/fletchc/tests_unsorted_process_linking_test/fletch.snapshot` returned 254, but we expected 0.

Details:
... the details from above ...
```

mit-mit, 2015-12-07:
We now print the exit code from the fletch exe as well. @mkustermann, is that enough?

mkustermann, 2015-12-10:
https://github.com/dart-lang/fletch/commit/50ffce48a0d44ac290c2623543a01d5051039c71 adds the `exitcode` information to all `CommandOutput`s and also indents the stdout/stderr/exitcode blocks to make it visually more obvious that there are two different commands we are running.

Marking this as fixed, though the error reporting could always be improved more.

closed 2015-12-10 by mkustermann