dart-archive/sdk #274: Some tests pass on LK (qemu) when the crash

kasperl, 2015-11-03:
See https://uberchromegw.corp.google.com/i/client.fletch/builders/fletch-lk-debug-arm-qemu/builds/114/steps/Test%20DebugLK-snapshot/logs/stdio

```
FAILED: fletchc-fletchvm debug_ia32 language/closures_initializer2_test
Expected: Crash RuntimeError 
Actual: Pass
CommandOutput[fletch_session]:

stdout:
Previous commands in this session:

$ out/DebugIA32/fletch create session 6 with .fletch-settings

 => 0
$ out/DebugIA32/fletch show log in session 6
[6: 2015-11-02 22:57:19.681 created session with file:///mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/.fletch-settings Settings(packages: file:///mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/.packages, options: [], constants: {}, device_address: null, device_type: DeviceType.mobile)]

 => 0
$ out/DebugIA32/fletch-vm &
$ out/DebugIA32/fletch attach tcp_socket 127.0.0.1:51835 in session 6

 => 0
$ out/DebugIA32/fletch export /mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/tests/language/closures_initializer2_test.dart to file /mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/out/DebugIA32/generated_compilations/fletchc/tests_language_closures_initializer2_test/fletch.snapshot in session 6
/mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/tests/language/closures_initializer2_test.dart:7:20:
Hint: Read `T` unhandled.
  A() : t = (() => T);
                   ^

 => 0
$ out/DebugIA32/fletch x-end session 6
Ended session '6'.

 => 0
$ out/DebugIA32/fletch-vm

stderr:
Fletch VM exitcode is 0

CommandOutput[vm]:

stdout:
Starting qemu...
Started with PID 390
Waiting for qemu to come up...
QEMU 2.4.0.1 monitor - type 'help' for more information
(qemu) Starting fletch...
Waiting for size...
Sending size (32237)...
Waiting for snapshot request...
Sending snapshot...
...............................
SNAPSHOT READ.
STARTING fletch-vm...
LOADING snapshot...
RUNNING program...

cpu 0 data abort, translation fault on read
DFAR 0x81907 (fault address)
DFSR 0x5 (fault status register)
halting
current_thread 0x8009c7c0, name Dart thread
r0  0x00081900 r1  0x80109208 r2  0x8010febb r3  0x80108070
r4  0x8009c6d0 r5  0x8010fde2 r6  0x80108070 r7  0x8010a129
r8  0x80109059 r9  0x00000001 r10 0x80114489 r11 0x80114481
r12 0x00000000 usp 0x00000000 ulr 0x00000000 pc  0x80060544
spsr 0x20000173
 usr r13 0x00000000 r14 0x00000000
 fiq r13 0x00000000 r14 0x00000000
 irq r13 0x00000000 r14 0x8005f654
asvc r13 0x8009d5a0 r14 0x8001104f
*svc r13 0x8009d678 r14 0x80046ddf
 und r13 0x00000000 r14 0x00000000
 sys r13 0x00000000 r14 0x00000000
bottom of stack at 0x8009d678:
0x8009d678: 8009d714 8009d70c 0000000a 8009d960 |............`...|
0x8009d688: 800991b8 800a43c0 8009c73c 00000000 |.....C..<.......|
0x8009d698: 80075260 80053941 00000020 bc11b5c3 |`R..A9.. .......|
0x8009d6a8: f8f6c51e 02a1e513 d817c230 c0c05cb9 |........0....\..|
0x8009d6b8: 14021fc4 5b711102 00000009 02025613 |......q[.....V..|
0x8009d6c8: 39002402 095a0006 13000000 00000000 |.$.9..Z.........|
0x8009d6d8: 011a0403 8009c6d0 00002842 8009c6d0 |........B(......|
0x8009d6e8: 00000000 8002c4eb 0215021f 011a0403 |................|
CRASH: starting debug shell... (reason = 9)
entering panic shell loop
Killing 390
! 
Killing 390

stderr:
qemu: terminating on signal 15 from pid 381
tools/lk/run_snapshot_lk_qemu.sh: line 84: /tmp/tmp.jDB6HQWsQZ/qemu.out: No such file or directory
tools/lk/run_snapshot_lk_qemu.sh: line 44: kill: (390) - No such process

Command[fletch_session]: 

There are three ways to reproduce this error:

  1. Run the test exactly as in this test framework. This is the hardest to
     debug using gdb:

    /mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/tools/testing/bin/linux/dart -c -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
       -Dfletch.version=`out/DebugIA32/fletch-vm --version` \
       tools/testing/dart/fletch_session_command.dart out/DebugIA32/fletch \
       /mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/tests/language/closures_initializer2_test.dart

  2. Run the helper program `tests/fletchc/run.dart` under `gdb` using
     `set follow-fork-mode child`. This can be confusing, but makes it
     easy to run a reproduction command in a loop:

    gdb -ex 'set follow-fork-mode child' -ex run --args \
        out/DebugIA32/dart -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
        -Dfletch.version=`out/DebugIA32/fletch-vm --version` \
        -c tests/fletchc/run.dart /mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/tests/language/closures_initializer2_test.dart

  3. Run the `fletch-vm` in gdb and attach to it via the helper program. This
     is the easiest way to debug using both gdb and lldb. You need to start two
     processes, each in their own terminal window:

    gdb -ex run --args out/DebugIA32/fletch-vm --port=54321

    out/DebugIA32/dart -Dtest.fletch_session_command.is_incremental_enabled=false -Dtest.fletch_settings_file_name=.fletch-settings \
      -Dfletch.version=`out/DebugIA32/fletch-vm --version` \
      -c -DattachToVm=54321 tests/fletchc/run.dart /mnt/data/b/build/slave/fletch-lk-debug-arm-qemu/build/fletch/tests/language/closures_initializer2_test.dart

Took 0:00:00.646777
```

skabet, 2015-11-03:
Should be fixed now.

kasperl, 2015-11-03:
I don't think this is fixed. Issue 273 is fixed, but this is about crashes being ignored on LK.

skabet, 2015-11-03:
Ahh, I see. I thought that it being ignored was a side-effect of the issue. I'll investigate this part further.

skabet, 2015-12-08:
This was fixed with the recent qemu roll.

closed 2015-12-08 by skabet