dart-archive/sdk #278: Resizing terminal when running tests leads to test failures

madsager, 2015-11-04:
I'm using a tiling window manager and when running tests I often open another terminal or do something else that rearranges my windows. That makes tests fail at this point. Peter pointed out to me that this is probably caused by delivery of SIGWINCH signals. That is correct. :)

We shouldn't treat SIGWINCH signals as fatal. If I add 

```c++
  if (signal == SIGWINCH) return;
```

in HandleSignal in driver/main.cc I can resize as desired, but I'm not sure that is the right level for this filtering?

@peter-ahe-google

peter-ahe-google, 2015-11-05:
@madsager is right about this not being the right level for filtering.

Looping in @wibling, as the agent is also affected.

This requires some amount of redesign:

* The immediate problem is that IsolateController.attachClient.handleCommand kills the isolate unconditionally when receiving a signal. It needs to know that some signals should be ignored.

* Another problem is the Fletch Agent which may incorrectly translate signals from Mac to Linux, for example, this is Linux: "18) SIGCONT 19) SIGSTOP 20) SIGTSTP", and this is Mac: "17) SIGSTOP 18) SIGTSTP 19) SIGCONT".

When the C++ client receives a signal, it needs to translate the signal number to something that is platform independent. This allows the persistent process to distinguish between signals that are normally ignored, and signals that should normally kill an application.

Once the persistent process has a platform-independent signal identifier, we can modify the agent to understand these new identifiers.

How should we identify signals? I suggest that we use the names from "kill -l", that is, strings, not numbers. Using numbers would be easy to confuse with the platform's signal numbers.

I don't think it is a high priority to support all sorts of signals, so I suggest that we temporarily limit all signal processing to SIGTERM and SIGINT.

peter-ahe-google, 2015-11-05:
[CL 1427363002](https://codereview.chromium.org/1427363002/) temporarily limits the signals that are intercepted by the C++ client.

wibling, 2015-11-05:
I wasn't aware of the difference in signal ids between mac and linux. We
should definitely fix that. Have you filed a bug about it?

peter-ahe-google, 2015-11-05:
I should probably close this bug when [CL 1427363002](https://codereview.chromium.org/1427363002/) lands. So I'll file another bug to track the problems I described above.

commit 83775f82 (2015-11-05): Temporarily limit which signals are intercepted

Fixes #278

R=ager@google.com, wibling@google.com

Review URL: https://codereview.chromium.org/1427363002 .
closed 2015-11-05 by peter-ahe-google