dart-archive/sdk #279: lib/async/timer_regress22626_test and lib/async/catch_errors12_test fails on arm

ricowind, 2015-11-04:
I will mark these in the status files for now
@skabet is this stuff that you did?

Full log at:
http://build.chromium.org/p/client.fletch/builders/target-fletch-linux-release-arm/builds/109/steps/test_ReleaseXARM%20failures/logs/stdio

Snippet from log:
stdout:
Previous commands in this session:

$ out/ReleaseXARM/fletch create session 1 with .fletch-settings

 => 0
$ out/ReleaseXARM/fletch show log in session 1
[1: 2015-11-04 13:34:47.866 created session with file:///b/build/slave/target-fletch-linux-release-arm/build/fletch/.fletch-settings Settings(packages: file:///b/build/slave/target-fletch-linux-release-arm/build/fletch/.packages, options: [], constants: {}, device_address: null, device_type: DeviceType.mobile)]

 => 0
$ out/ReleaseXARM/fletch-vm &
$ out/ReleaseXARM/fletch attach tcp_socket 127.0.0.1:40733 in session 1

 => 0
$ out/ReleaseXARM/fletch run /b/build/slave/target-fletch-linux-release-arm/build/fletch/tests/lib/async/timer_regress22626_test.dart in session 1
Uncaught exception: Instance of 'ExpectException' {
  ExpectException.message: 'Expect.fail('should have been cancelled by now')'
}
Stack trace:
> 0: Expect._fail                    @new ExpectException(message)
  1: Expect.fail                     @_fail("Expect.fail('${msg}')")
  2: call                            @Expect.fail("should have been cancelled by now")

/b/build/slave/target-fletch-linux-release-arm/build/fletch/third_party/dart/pkg/expect/lib/expect.dart:385:11
380  
381    static String _getMessage(String reason)
382        => (reason == null) ? "" : ", '$reason'";
383  
384    static void _fail(String message) {
385      throw new ExpectException(message);
386    }
387  }
388  
389  bool _identical(a, b) => identical(a, b);
390  

 => 255
$ out/ReleaseXARM/fletch x-end session 1
Ended session '1'.

 => 0
$ out/ReleaseXARM/fletch-vm

stderr:
Fletch VM exitcode is 255

ricowind, 2015-11-05:
async/catch_errors12_test is flaky

mkustermann, 2015-11-25:
The `lib/async/timer_regress22626_test` test seems to be passing now (arm and arm+lk) after 2d9a62a8c5512b822eb886024f25098902b4c7f1 landed.

mit-mit, 2015-12-07:
@mkustermann, can we close the bug then, or is there anything else to do?

closed 2015-12-08 by mkustermann