dart-archive/sdk #291: Assertion error in VM.

sigurdm, 2015-11-06:
To reproduce, replace FletchNoSuchMethodError.toString in lib/system/nsm.dart with:
```
  String toString()  => 'FletchNoSuchMethodError($_receiver, $_receiverClass, selector: $_receiverSelector)';
```
And run
```
main(args) {
 try {
   new Object().fisk();
 } catch (e) {
   print(e);
 }
}
```
With the debug build of fletch.
Output:
```
Warning: No method named 'fisk' in class 'Object'.
   new Object().fisk();
   ^^^^^^^^^^^^^^^^^^^
../../src/vm/object.h:1553: error: expected: object->IsArray()
fish: Job 1, “out/DebugX64/fletch run ~/test/main.dart” terminated by signal SIGABRT (Abort)
```

mit-mit, 2015-12-07:
@skabet is this relevant?

skabet, 2015-12-07:
@sigurdm, can you confirm we fixed it that day?

sigurdm, 2015-12-08:
I think we found out that printing _receiverClass crashes the fletch vm. I am not sure if that is considered ok because it cannot happen from user code.

skabet, 2015-12-08:
Yes, that was it. And yes, that is ok. We haven't implemented that yet, and it's not exposed to the user. Closing.

closed 2015-12-08 by skabet