dart-archive/sdk #296: NativeProcessSpawnDetached seems to generate coredumps on BB

mkustermann, 2015-11-06:
We have currently (despite green buildbot) coredumps archived. For example for [this build](https://build.chromium.org/p/client.fletch/builders/fletch-linux-release-x86/builds/172/steps/Test%20ReleaseX64Clang/logs/coredumps) we have a coredump. 

Looking at one of the coredumps with gdb, the stack trace gives:

```
(gdb) bt
#0  0x00007fef84abfcc9 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007fef84ac30d8 in __GI_abort () at abort.c:89
#2  0x00000000004231e6 in fletch::Platform::ImmediateAbort() ()
#3  0x00000000004213bd in fletch::DynamicAssertionHelper::Fail(char const*, ...) ()
#4  0x000000000040d2ad in Native_NativeProcessSpawnDetached ()
#5  0x000000000040942b in fletch::Engine::Interpret(fletch::TargetYieldResult*) ()
#6  0x000000000040c1e0 in fletch::Interpreter::Run() ()
#7  0x0000000000419a80 in fletch::Scheduler::InterpretProcess(fletch::Process*, fletch::Heap*, fletch::ThreadState*, bool*) ()
#8  0x00000000004198ba in fletch::Scheduler::RunInterpreterLoop(fletch::ThreadState*) ()
#9  0x00000000004196e8 in fletch::Scheduler::RunInThread() ()
#10 0x00000000004205ce in fletch::ThreadPool::RunThread(void*) ()
#11 0x00007fef8587a182 in start_thread (arg=0x7fef7ffff700) at pthread_create.c:312
#12 0x00007fef84b8347d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

@wibling  The `Native_NativeProcessSpawnDetached` could mean this is related to your work on the agent. Could you take a look at this? Any ideas?

wibling, 2015-11-09:
Sure. I will take a look. I don't have any ideas off hand what might be the
issue.

/gustav

wibling, 2015-12-07:
I believe this is fixed as of 5667cae6ec841d6a06da4ffcbb1cc7080d0f7cfe.

closed 2015-12-07 by wibling